GloDroid/glodroid_manifest #1: [ Request ] Android tv os + ir remote

hostkit, 2020-02-01:
Please add Android tv os and ir remote

hostkit, 2020-04-26:
Please building Android tv img please

herbsmn, 2020-11-17:
It looks like ir support was added to the orange-pi lite 2 and orange pi 3 in this 5.4 kernel just a week ago:
https://github.com/orangepi-xunlong/linux-orangepi/commit/5bf088818e4e76dacd3ae58ba1549b80bd679546
https://github.com/orangepi-xunlong/linux-orangepi/commit/7f272478bc2c5aff0dade7c9b8bf3cfa0ed620db

herbsmn, 2020-12-01:
There was a small bit of documentation in this depreciated wiki page about building GloDroid with AndroidTV: https://github.com/GloDroid/glodroid_manifest/wiki/Getting-started#android-tv

rsglobal, 2021-05-12:
Android TV doesn't make any sense with software codecs. This issue is blocked by it at this moment.