GloDroid/glodroid_manifest #5: build problem

tantaliom1985, 2020-02-14:
/home/admin/GloDroid/build/blueprint/pathtools/glob.go:166 +0x455
github.com/google/blueprint/pathtools.startGlob(0xaf7260, 0xf18568, 0xc0041420c0, 0x12, 0x0, 0x0, 0x0, 0xc004142001, 0xda00000000000020, 0x0, ...)
        /home/admin/GloDroid/build/blueprint/pathtools/glob.go:52 +0xb2
github.com/google/blueprint/pathtools.osFs.Glob(...)
        /home/admin/GloDroid/build/blueprint/pathtools/fs.go:154
github.com/google/blueprint.(*Context).glob(0xc0000989c0, 0xc0041420c0, 0x12, 0x0, 0x0, 0x0, 0x2a2a2f6b64732f73, 0x2a2f, 0x0, 0xc004142080, ...)
        /home/admin/GloDroid/build/blueprint/glob.go:64 +0x217
github.com/google/blueprint.(*baseModuleContext).GlobWithDeps(0xc00e76fd40, 0xc0041420c0, 0x12, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, ...)
        /home/admin/GloDroid/build/blueprint/module_ctx.go:277 +0x6d
android/soong/android.(*androidModuleContext).GlobFiles(0xc080cbe340, 0xc0041420c0, 0x12, 0x0, 0x0, 0x0, 0x12, 0xc000076300, 0xc036b41218)
        /home/admin/GloDroid/build/soong/android/module.go:1504 +0x7e
android/soong/android.expandOneSrcPath(0xb0ca20, 0xc080cbe340, 0xc036b41218, 0x4, 0x0, 0x0, 0x0, 0xc000076300, 0x0, 0x0, ...)
        /home/admin/GloDroid/build/soong/android/paths.go:330 +0xd99
android/soong/android.PathsAndMissingDepsForModuleSrcExcludes(0xb0ca20, 0xc080cbe340, 0xc080ff9790, 0x1, 0x1, 0x0, 0x0, 0x0, 0x6, 0x6, ...)
        /home/admin/GloDroid/build/soong/android/paths.go:289 +0x446
android/soong/android.PathsForModuleSrcExcludes(0xb0ca20, 0xc080cbe340, 0xc03141b790, 0x1, 0x1, 0x0, 0x0, 0x0, 0xc000076300, 0x0, ...)
        /home/admin/GloDroid/build/soong/android/paths.go:236 +0x9b
android/soong/android.PathsForModuleSrc(...)
        /home/admin/GloDroid/build/soong/android/paths.go:226
android/soong/java.(*ExportedDroiddocDir).GenerateAndroidBuildActions(0xc043133100, 0xb0ca20, 0xc080cbe340)
        /home/admin/GloDroid/build/soong/java/droiddoc.go:1818 +0x1af
android/soong/android.(*ModuleBase).GenerateBuildActions(0xc043133100, 0xb0ae00, 0xc00e76fd40)
        /home/admin/GloDroid/build/soong/android/module.go:860 +0x706
github.com/google/blueprint.(*Context).generateModuleBuildActions.func2.1(0xc006876f00, 0xc00e76fd40)
        /home/admin/GloDroid/build/blueprint/context.go:2303 +0x80
github.com/google/blueprint.(*Context).generateModuleBuildActions.func2(0xc006876f00, 0x0)
        /home/admin/GloDroid/build/blueprint/context.go:2304 +0x339
github.com/google/blueprint.(*Context).parallelVisit.func1.1(0xc080c200c0, 0xc006876f00, 0xc00006e9c0, 0xc00006e960)
        /home/admin/GloDroid/build/blueprint/context.go:1709 +0x30
created by github.com/google/blueprint.(*Context).parallelVisit.func1
        /home/admin/GloDroid/build/blueprint/context.go:1708 +0x124
10:01:03 soong bootstrap failed with: exit status 1
ninja: build stopped: subcommand failed.build/make/core/main.mk:21: recipe for target 'run_soong_ui' failed
make: *** [run_soong_ui] Error 1

rsglobal, 2020-02-15:
Could you try with v0.1.0?  
https://github.com/glodroid/glodroid_manifest/tree/v0.1.0

```
mkdir -p GloDroid
cd GloDroid
repo init -u https://github.com/glodroid/glodroid_manifest -b refs/tags/v0.1.0
repo sync -cq
source ./build/envsetup.sh
lunch plus2e-userdebug
make sdcard 2>&1 | tee build.log
```

rsglobal, 2020-04-22:
@AndriiZ ,

Thanks for your bug report please try to install `dtc` and build again:
```sudo apt-get install -y device-tree-compiler```

AndriiZ, 2020-04-22:
@rsglobal thanks
Now another error
[ 42% 252/597] including device/glodroid/common/Android.mk ...
FAILED:
build/make/core/base_rules.mk:325: error: device/generic/goldfish/gatekeeper: MODULE.TARGET.SHARED_LIBRARIES.gatekeeper.ranchu already defined by device/generic/goldfish/gatekeeper.
08:30:45 ckati failed with: exit status 1
build/make/core/main.mk:21: recipe for target 'run_soong_ui' failed

rsglobal, 2020-04-22:
It looks like issue in environment. Could you close your terminal and start from `source build/envsetup.sh` etc...

rsglobal, 2020-06-23:
Can not reproduce. Please re-open issue if it is still valid.

closed 2020-06-23 by rsglobal

bigc0127, 2020-07-10:
build/make/core/base_rules.mk:325: error: device/generic/goldfish/gatekeeper: MODULE.TARGET.SHARED_LIBRARIES.gatekeeper.ranchu already defined by device/generic/goldfish/gatekeeper

I have this same error

pratik0007, 2023-09-15:
i got the same error. i did try installing this 'sudo apt-get install -y device-tree-compiler'  but so satisfactory result. how did you guys fixed it ? 
[log.txt](https://github.com/GloDroid/glodroid_manifest/files/12623053/log.txt)