GloDroid/glodroid_manifest #7: Script to install on emmc

AndriiZ, 2020-02-17:
It is possible to install Armbian to emmc https://docs.armbian.com/User-Guide_Getting-Started/#how-to-install-to-emmc-nand-sata-usb
It is also possible to burn LibreELEC image to emmc for LibreELEC 
I think, it is possible to burn Android to emmc, but probably it will require some extra work.

rsglobal, 2020-02-19:
I hope to fix u-boot fastboot gadget driver. After that we could use classic way to flash Android to emmc.

hostkit, 2020-02-20:
have you ever thought of using dd command to install to emmc? Maybe working

rsglobal, 2020-03-10:
Changes in the next branch contains all necessary adjustments to enable boot from EMMC.
But for flashing you have to use dd (from armbian image on SDCARD and Android on USB for example).

hadimz, 2020-04-08:
Can you please elaborate how we can install using Armbian on SD card?
I tried booting from SD while having the android img file on a usb stick. I used dd command to write the image to emmc but it didn't boot when I removed the SD card and restarted.
Do I need to use another tool than dd or did I do something wrong?

AndriiZ, 2020-04-24:
Provided binary image for OPi3 large than internal emmc, so burn with dd fail. Subsequent boot fails.

hadimz, 2020-05-22:
can this version be installed on the internal storage of orange pi plus 2e?

rsglobal, 2020-05-22:
eMMC works on plus2e and opi3 now. You can build next branch to try it. 
I will post release soon.

closed 2020-06-23 by rsglobal