GloDroid/glodroid_manifest #8: [ Request ] root access + playstore

hostkit, 2020-02-18:
Please add playstore dan root access in this rom

rsglobal, 2020-02-18:
Playstore is a part of google suite that are only accessible for mobile brands.
Root access is enabled on userdebug builds, and this project is currently supports userdebug only, so it is already enabled.

closed 2020-02-18 by rsglobal

hostkit, 2020-02-19:
How about playstore tv? Isn't GooglePlay available for TV devices?

rsglobal, 2020-02-19:
```
Add GMS to your devices
While the Android Open Source Project (AOSP) provides common, device-level functionalities such 
as email and calling, GMS is not part of AOSP. GMS is only available through a license with Google
and delivers a holistic set of popular apps and cloud-based services. Note that GMS apps can vary
based on country availability and requirements.
```

You can learn more at:
https://www.android.com/gms/

hostkit, 2020-02-20:
So we cant install gapps pre install? Especially opengapps