GloDroid/glodroid_manifest #9: [Bug] glodroid 0.1.0

hostkit, 2020-02-21:
After flashing your release sdcard image and load my sdcard. My opi only stuck on Android logo more than 1hours. I don't know to get log.

rsglobal, 2020-02-21:
I suppose that you can have no heat-sink mounted on your OPI SOC.

We are using 40x40mm heat-sink.

[image]

hostkit, 2020-02-21:
My opi mount mini heatsink and fan

hostkit, 2020-02-21:
> I suppose that you can have no heat-sink mounted on your OPI SOC.
> 
> We are using 40x40mm heat-sink.
> 
> [image]

I think your device not plus2e or maybe opiplus2e have different type?
[image]

rsglobal, 2020-02-22:
Yes, you are right, I sent a photo of opi-win (A64 soc). 
We have same size heatsink on opi-plus2e. 

If you are sure that there is no overheat, please check you have got 5V, 2A power supply and cable with low voltage drop.

hostkit, 2020-02-22:
> Yes, you are right, I sent a photo of opi-win (A64 soc).
> We have same size heatsink on opi-plus2e.
> 
> If you are sure that there is no overheat, please check you have got 5V, 2A power supply and cable with low voltage drop.

I don't know how to check when board offline, but when using libreelec only 45-47.
Adapter? Ok will be check my volt and Amper

rsglobal, 2020-03-05:
According to my experience the heat-sink on your photo is too small when using OPI's GPU.
Such distributive as armbian does not use GPU hardware acceleration and can work even without heat-sink.
I believe libreelec also.
Soon we plan to switch to kernel 5.6, where thermal driver is available. After that you should experience just performance drop instead of hang in case of overheating.

hostkit, 2020-03-10:
Do you know, how to remove old heatsink with thermal tape? I want to replace with new heatsink

hostkit, 2020-03-10:
Ok. I have been successful remove old heatsink. which part must be installed new 40x40 heatsinks? Part 1-5 single heatsink 40x40 without part 6 or 1-5 single heatsink+ part 6 with small heatsink?
[image]

rsglobal, 2020-03-10:
SOC heats mostly. So only 5.

rsglobal, 2020-04-23:
Try v0.2.0. For me it works even without heatsink:
https://github.com/GloDroid/glodroid_manifest/releases/tag/v0.2.0

closed 2020-05-10 by hostkit