GloDroid/glodroid_manifest #10: Audio/Video/ADB

YevheniL, 2020-03-06:
Hello, could you please answer on few questions:
1) How to connect by adb via usb wire? For now when i plug it i can not see any activity. And command adb devices returning empty result. 
2) What happen with video playing ? In the first your release it is worked, but for now not. How to return this possibility?

rsglobal, 2020-03-06:
Hi,

ADB should work. Youtube video in webview browser should also work. 
Which release and board are you using?
Are you using prebuilt image or compile from scratch?

YevheniL, 2020-03-06:
In Youtube it is working fine, video from the folder directory not. 
For now i using the last one version from this repository.
ADB via Ethernet or via USB ? 
I had error on the device Failed to get adbd socket, No such file or directory. 
Thank you very match for the fast answer !

YevheniL, 2020-03-06:
It is write, that i am not superuser, that`s why adb is stopped.

rsglobal, 2020-06-23:
To make adb / fastboot work without sudo you have to install adb package:
```sudo apt install adb```

Video playback issue is tracked in #13

closed 2020-06-23 by rsglobal