GloDroid/glodroid_manifest #13: Provide support VP8-9 video codecs

MaksymPrymierov, 2020-03-16:


rsglobal, 2020-03-16:
Codec2 should already include VP8 and VP9.

Sunxi VPU have only h264 and h265 codecs driver.

joek85, 2020-04-16:
`2020-04-13 09:44:34.596 1905-1905/com.android.gallery3d D/StrictMode: StrictMode policy violation: android.os.strictmode.ContentUriWithoutPermissionViolation: content://media/external/video/media/93 exposed beyond app through Intent.getData() without permission grant flags; did you forget FLAG_GRANT_READ_URI_PERMISSION?
        at android.os.StrictMode.onContentUriWithoutPermission(StrictMode.java:2091)
        at android.net.Uri.checkContentUriWithoutPermission(Uri.java:2400)
        at android.content.Intent.prepareToLeaveProcess(Intent.java:10803)
        at android.content.Intent.prepareToLeaveProcess(Intent.java:10744)
        at android.app.Instrumentation.execStartActivity(Instrumentation.java:1710)
        at android.app.Activity.startActivityForResult(Activity.java:5211)
        at android.app.Activity.startActivityForResult(Activity.java:5169)
        at com.android.gallery3d.app.PhotoPage.playVideo(PhotoPage.java:1229)
        at com.android.gallery3d.app.PhotoPage.onSingleTapUp(PhotoPage.java:1157)
        at com.android.gallery3d.ui.PhotoView$MyGestureListener.onSingleTapUp(PhotoView.java:1012)
        at com.android.gallery3d.ui.GestureRecognizer$MyGestureListener.onSingleTapUp(GestureRecognizer.java:79)
        at android.view.GestureDetector.onTouchEvent(GestureDetector.java:730)
        at com.android.gallery3d.ui.GestureRecognizer.onTouchEvent(GestureRecognizer.java:58)
        at com.android.gallery3d.ui.PhotoView.onTouch(PhotoView.java:941)
        at com.android.gallery3d.ui.GLView.dispatchTouchEvent(GLView.java:341)
        at com.android.gallery3d.ui.GLView.dispatchTouchEvent(GLView.java:302)
        at com.android.gallery3d.ui.GLView.dispatchTouchEvent(GLView.java:322)
        at com.android.gallery3d.ui.GLRootView.dispatchTouchEvent(GLRootView.java:479)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3060)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2755)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3060)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2755)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3060)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2755)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3060)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2755)
        at com.android.internal.policy.DecorView.superDispatchTouchEvent(DecorView.java:465)
        at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1849)
        at android.app.Activity.dispatchTouchEvent(Activity.java:4012)
        at com.android.internal.policy.DecorView.dispatchTouchEvent(DecorView.java:423)
        at android.view.View.dispatchPointerEvent(View.java:14212)
        at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:5652)
        at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:5455)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4958)
        at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:5011)
        at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4977)
        at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:5117)
        at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4985)
        at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:5174)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4958)
        at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:5011)
        at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4977)
        at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4985)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4958)
        at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:7675)
        at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:7644)
        at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:7605)
        at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:7800)
        at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:188)
        at android.os.MessageQueue.nativePollOnce(Native Method)
    	at android.os.Message
2020-04-13 09:44:34.605 368-2060/system_process I/ActivityTaskManager: START u0 {act=android.intent.action.VIEW dat=content://media/external/video/media/93 typ=video/* cmp=com.android.gallery3d/.app.MovieActivity (has extras)} from uid 10086
2020-04-13 09:44:34.612 368-389/system_process I/DropBoxManagerService: add tag=system_app_strictmode isTagEnabled=true flags=0x2
2020-04-13 09:44:34.628 1905-1905/com.android.gallery3d W/OrientationEventListener: Cannot detect sensors. Invalid disable
2020-04-13 09:44:34.655 1905-1905/com.android.gallery3d W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@83542cc
2020-04-13 09:44:34.718 1905-1905/com.android.gallery3d W/AudioManager: Use of stream types is deprecated for operations other than volume control
2020-04-13 09:44:34.718 1905-1905/com.android.gallery3d W/AudioManager: See the documentation of requestAudioFocus() for what to use instead with android.media.AudioAttributes to qualify your playback use case
[... cut ...]
2020-04-13 09:44:37.455 368-2480/system_process E/InputDispatcher: Window handle nav_input_consumer has no registered input channel
2020-04-13 09:44:37.473 368-2480/system_process E/InputDispatcher: Window handle nav_input_consumer has no registered input channel
2020-04-13 09:44:37.486 368-2060/system_process E/InputDispatcher: Window handle nav_input_consumer has no registered input channel
2020-04-13 09:44:38.223 368-1311/system_process E/TaskPersister: File error accessing recents directory (directory doesn't exist?).`

hello, i am trying to play a video and it cannot be played.

rsglobal, 2020-06-23:
Need to check if https://github.com/robherring/gbm_gralloc/pull/15 fixes this issue.

rsglobal, 2020-06-24:
After applying https://github.com/robherring/gbm_gralloc/pull/15, video playback is working.
I will include it into the next release.

closed 2020-06-24 by rsglobal