GloDroid/glodroid_manifest #14: Sdcard issue

joek85, 2020-04-06:
hello, i just downloaded the sdcard_trunc.img.gz file from https://github.com/GloDroid/glodroid_manifest/releases/tag/v0.1.0
and extracted via gunzip, i write it via 
`dd if=sdcard_trunc.img of=/dev/sdd bs=4k count=300000`
and also tried it with "disk image writer" that comes out from ubuntu.
i have 3 Apecer 8GB class 10 sdcards, formatted with Gparted to Ext4.
after more than 20 tries the board didn't boot at all.
it just booted once and the green led turned on, booted into Android boot animation and stuck there.
i don't have a serial to usb yet to debug it but from the adb logs it showed a failure at "service gatekeeper cannot started" and it looped.
now that error i faced it earlier from my custom Android 7 image and it turned out that the issue was some fex entries wasn't compatible with the board which is Plus2E, i think it was related to the mmc i don't remember exactly.
do you have another img to test? maybe the GloDroid preview v4 img that is missing.
thank you.

rsglobal, 2020-04-15:
Hello,

Have you run `sync` command after dd?

joek85, 2020-04-16:
hello,
yes i did, however i ended up downloading the whole source, and compiled, deployed successfully.
how can i change the screen resolution? currently it is 800x480, i want to scale it up a bit maybe 1280x720 or 1080p.
thank you.

rsglobal, 2020-04-23:
Screen resolution is set based on your display capabilities that are read as EDID blob.
Currently AOSP have an internal issue that makes it impossible to distinguish between interlaced and not interlaced mode, therefore drm_hwcomposer have disabled all modes with interlaced flag.
I suppose your display supports only 800x480 without interlacing.

rsglobal, 2020-06-23:
GloDroid switched to different flashing procedure that is now available in next branch, and will be available in next release.

closed 2020-06-23 by rsglobal