GloDroid/glodroid_manifest #15: Porting to Orange Pi Plus / Plus 2

vikitinho, 2020-04-14:
I'm trying to port your project to run on an Orange Pi Plus / Plus 2 and thus contribute to more users.

I created a new device dir based on the plus2e configuration files, changed the dtb kernel to the respective one for this device (sun8i-h3-orangepi-plus.dtb), did the same for the uboot configuration.

After generating the sdcard.img, android starts, but I see the TV message "Android System: There's an internal problem with your device. Contact your manufacturer for details."

Attached ttyUSB log.

Is there anyone who can and wants to help me?
[minicom-09042020.txt](https://github.com/GloDroid/glodroid_manifest/files/4477362/minicom-09042020.txt)

rsglobal, 2020-04-15:
Hello,

Is seems that your system have similar issue with opi3 and musb driver is configured to host mode only. The solution was to override this settings using device-tree overlay. Refer to https://github.com/GloDroid/glodroid_device/commit/a89844656b7343628b5892469b18ccc10e68c2a5 for more information.

vikitinho, 2020-04-15:
I tried to make the suggested change, but I get a new error ...

```
Loading Environment from EXT4... 
** Unable to use mmc 0:auto for loading the env **
In:    serial
Out:   serial
Err:   serial
Net:   phy interface7
eth0: ethernet@1c30000
starting USB...
No working controllers found
Autoboot in 1 seconds, press <Space> to stop
1871 bytes read in 1 ms (1.8 MiB/s)
## Executing script at 53000000
switch to partitions #0, OK
mmc0 is current device

MMC read: dev # 0, block # 8192, count 49152 ... 49152 blocks read: OK
failed on fdt_overlay_apply(): FDT_ERR_NOTFOUND
## Booting Android Image at 0x50008000 ...
Kernel load addr 0x40088000 size 7207 KiB
Kernel command line: androidboot.hardware=plus buildvariant=userdebug
RAM disk load addr 0x43300000 size 6122 KiB
   Loading Kernel Image
   Loading Ramdisk to 49a05000, end 49fff415 ... OK

Starting kernel ...

Error: invalid dtb and unrecognized/unsupported machine ID
  r1=0x00000000, r2=0x40000100
  r2[]=05 00 00 00 01 00 41 54 00 00 00 00 00 00 00 00
Available machine support:

ID (hex)        NAME
ffffffff        Generic DT based system
ffffffff        Allwinner suniv Family
ffffffff        Allwinner sun9i Family
ffffffff        Allwinner A83t board
ffffffff        Allwinner sun8i Family
ffffffff        Allwinner sun7i (A20) Family
ffffffff        Allwinner sun6i (A31) Family
ffffffff        Allwinner sun4i/sun5i Families

Please check your kernel config and/or bootloader.
```

'll do more detailed tests and I'll be back when I have more information. I appreciate the support.

rsglobal, 2020-04-16:
Could you share you changes?

hostkit, 2020-04-26:
> Hello,
> 
> Is seems that your system have similar issue with opi3 and musb driver is configured to host mode only. The solution was to override this settings using device-tree overlay. Refer to [GloDroid/glodroid_device@a898446](https://github.com/GloDroid/glodroid_device/commit/a89844656b7343628b5892469b18ccc10e68c2a5) for more information.
> 
> At least you should be able to run `adb shell`, use `fastbootd`, etc. after that.

Sometimes plus2e Android System: There's an internal problem with your device. Contact your manufacturer for details.

BigJacek, 2020-06-01:
Hello,
Clould you pls. extend the project to support Orange Pi Plus / Plus 2?
Best regards,
Jacek

rsglobal, 2020-06-02:
Hello Jacek,

Thank you for your request. 
Currently our team does not have such boards.
I will ask Xunlong to sponsor our project with boards, and will start working on new devices as soon as they arrive.

BigJacek, 2020-06-02:
Hello,
Thank you very much for your quick answer. 
I keep my fingers crossed for a quick delivery of the device from your sponsor and I hope that Orange Pi Plus / Plus 2 will soon be supported.
Best regards,
Jacek

BigJacek, 2020-06-05:
Btw. Can I help in some way?
I'm new here, but if there's anything I can do, I'd love to join in.

rsglobal, 2020-06-05:
Hello Jacek,

Yes you can.
Enabling basic functionality for plus / plus2 is trivial task, but it require testing on real hardware.
You can build it yourself or I can provide images for testing.

Regards,
Roman

BigJacek, 2020-06-05:
OK.
So I have one piece of Orange Pi Plus2 H3 and I would like to test something.
So let's start with an image for testing.
After all, I would like to be able to build the image myself.

BigJacek, 2020-06-08:
Hello Roman,
Can you provide images for testing?
Regards,
Jacek

rsglobal, 2020-06-08:
Hi Jacek,
Just give me a couple of days.
Currently I am working on rpi4 integration. I will start working on this issue soon.
Regards,
Roman

BigJacek, 2020-06-08:
Roman,
Sure, no problem.
In the meantime I'm trying  to build GloDroid repo on my PC.
We're in contact.
Jacek

BigJacek, 2020-06-11:
Hello Roman,
I'm trying to build GloDroid repo on my PC based on description in the web.
There are some problems and finally there are unable to sync repo.
Pls. see errors below:
error: build/blueprint/: platform/build/blueprint checkout 675d72d5f3b3200f6dcc174d8bda00c227abd36c 
error: external/autotest/: platform/external/autotest checkout 0c09d81ab60ad4e45c0dacb9f4ddb989f16e20e6 
error: external/kmod/: platform/external/kmod checkout e6417e44d57fcdfcce1eb96f9443dca49bdf91ea 
error: external/libunwind/: platform/external/libunwind checkout 7da670a13886265d84bd5d78475a6329c038df10 
and finally:
error: unable to create file tests/P_str_escape/str\\escape.rs: Zły argumentdating files:  57% (491/856)
error: Cannot checkout platform/frameworks/compile/slang: GitError: Cannot initialize work tree for platform/frameworks/compile/slang
error.GitError: Cannot initialize work tree for platform/frameworks/compile/slang

Is there any solution for that?
Jacek

rsglobal, 2020-06-11:
Hello Jacek,

Could you check disk space usage? 
`df -h`

BigJacek, 2020-06-11:
Hello,
Disk usage:
System plików  rozm. użyte dost. %uż. zamont. na
udev            1,9G     0  1,9G   0% /dev
tmpfs           391M  1,9M  390M   1% /run
/dev/sda5        20G   14G  4,6G  75% /
tmpfs           2,0G     0  2,0G   0% /dev/shm
tmpfs           5,0M  4,0K  5,0M   1% /run/lock
tmpfs           2,0G     0  2,0G   0% /sys/fs/cgroup
/dev/loop0       55M   55M     0 100% /snap/core18/1705
/dev/loop1       55M   55M     0 100% /snap/core18/1754
/dev/loop3       50M   50M     0 100% /snap/snap-store/433
/dev/loop2      256M  256M     0 100% /snap/gnome-3-34-1804/36
/dev/loop4       63M   63M     0 100% /snap/gtk-common-themes/1506
/dev/loop5       31M   31M     0 100% /snap/snapd/7777
/dev/loop6       28M   28M     0 100% /snap/snapd/7264
/dev/loop7       50M   50M     0 100% /snap/snap-store/454
/dev/loop8      241M  241M     0 100% /snap/gnome-3-34-1804/24
/dev/sda1       511M  4,0K  511M   1% /boot/efi
vmhgfs-fuse     238G  186G   53G  78% /mnt/hgfs
tmpfs           391M   20K  391M   1% /run/user/1000
/dev/sdb1       299G   69G  230G  23% /media/adminek/Citrix

repo is on resource sdb1

Regards,
Jacek

rsglobal, 2020-06-12:
I've had similar issue.
In my case repository in `.repo/projects/ `directory was broken.
Try to delete repositories listed in error log you've attached from .repo/projects/ and run `repo sync` again.

BigJacek, 2020-06-12:
Yes. I thought about it too, and I did, but still no luck.
I removed the whole repo and I'm starting over.

BigJacek, 2020-06-12:
The fourth time after each repo removal, the synchronisation was finally completed.

BigJacek, 2020-06-12:
Hello,
I wanted to biuld image for sdcard (for testing purposes) but everytime I've got an error:
$ make sdcard
============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=10
TARGET_PRODUCT=opi_pc
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_ARCH=arm
TARGET_ARCH_VARIANT=armv7-a-neon
TARGET_CPU_VARIANT=generic
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-5.4.0-37-generic-x86_64-Ubuntu-20.04-LTS
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=QQ2A.200305.004.A1
OUT_DIR=out
============================================
[100% 47/47] out/soong/.bootstrap/bin/soong_build out/soong/build.ninja
FAILED: out/soong/build.ninja
out/soong/.bootstrap/bin/soong_build -t -l out/.module_paths/Android.bp.list -b out/soong -n out -d out/soong/build.ninja.d -globFile out/soong/.bootstrap/build-globs.ninja -o out/soong/build.ninja Android.bp
Killed
21:26:56 soong bootstrap failed with: exit status 1
ninja: build stopped: subcommand failed.
#### failed to build some targets (05:19 (mm:ss)) ####
What could be wrong?

Jacek

BigJacek, 2020-06-12:
Hello,
I wanted to biuld image for sdcard (for testing purposes) but everytime I've got an error:
$ make sdcard
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=10
TARGET_PRODUCT=opi_pc
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_ARCH=arm
TARGET_ARCH_VARIANT=armv7-a-neon
TARGET_CPU_VARIANT=generic
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-5.4.0-37-generic-x86_64-Ubuntu-20.04-LTS
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=QQ2A.200305.004.A1
OUT_DIR=out
[100% 47/47] out/soong/.bootstrap/bin/soong_build out/soong/build.ninja
FAILED: out/soong/build.ninja
out/soong/.bootstrap/bin/soong_build -t -l out/.module_paths/Android.bp.list -b out/soong -n out -d out/soong/build.ninja.d -globFile out/soong/.bootstrap/build-globs.ninja -o out/soong/build.ninja Android.bp
Killed
21:26:56 soong bootstrap failed with: exit status 1
ninja: build stopped: subcommand failed.

failed to build some targets (05:19 (mm:ss))
What could be wrong?

Jacek

rsglobal, 2020-06-12:
You are out of memory.
Close chrome and other apps and try again. Also you should have at least 16gb of RAM.
Doing make -j1 sdcard could also help, but on final building stages.

BigJacek, 2020-06-12:
Ah, ok.
I'm working on VM so I just can forget about this.

hostkit, 2020-06-13:
> You are out of memory.
> Close chrome and other apps and try again. Also you should have at least 16gb of RAM.
> Doing make -j1 sdcard could also help, but on final building stages.

Higher specific unable build it self

BigJacek, 2020-06-13:
Lack of RAM.
I won't go through with it now.

BigJacek, 2020-06-14:
Hello Roman,
Can you provide some images for testing (when you have a moment, of course)?
Regards,
Jacek

BigJacek, 2020-06-14:
Unfortunately, that's all I can do right now.

rsglobal, 2020-06-18:
Hello Jacek,

I've build images for you. I expect some issues since kernel is mainline without Google patches. But UI should start.
https://drive.google.com/file/d/1gJCYklek2yMQDqpokeZo2pl9ILPRUtVK/view?usp=sharing

BigJacek, 2020-06-18:
Hello Roman,
I'm starting to work on this.
JaS

BigJacek, 2020-06-18:
Roman,
I've got a problem.
In recovery mode device is not listed by command "adb devices" (every time I get a message: error: no devices/emulators found).
However device is mapped as removables device (Onda MT65xx Preloader).
I'm using Ubuntu 20.04 LTS.
Regards,
Jacek

vikitinho, 2020-06-18:
Hi Roman,  

I did the same test and got the same error.

Attached ttyUSB log.

[minicom-18062020.log](https://github.com/GloDroid/glodroid_manifest/files/4800956/minicon-18062020.log)

rsglobal, 2020-06-18:
In case you are in fastbootd mode, adb won't work.

I'll look into it tomorrow.

BigJacek, 2020-06-18:
I did this also but no reaction on Pi Plus 2 side. (not reload/reboot)

BigJacek, 2020-06-18:
< waiting for any device >

and nothing more...

rsglobal, 2020-06-19:
Jacek,

Unfortunately I can see no usb_otg node in orange pi plus device tree in both u-boot and kernel project.
You can try to copy-paste it from opi pc, recompile kernel and try again.

rsglobal, 2020-06-19:
Plus2 board have no schematic publicly available.
To enable USB I should specify to which H3 SOC pin USB-ID OTG contact is routed.

Can anybody who have broken board, remove SOC and probe ID pin using multimeter?

BigJacek, 2020-06-19:
I dont know if it will be usefull/helpful but there is orangepi-plus-h3-v1_1_schematic available (https://pan.baidu.com/s/1bnppzv5).
As far as I can see, the firmware is the same for both versions, so maybe the architecture is similar.

rsglobal, 2020-06-19:
Thank you for the link.
Lets assume plus and plus2 have same pcb layout.
From the circuit diagram:
```
PG12 - USB0_IDDET
PL2 - USB0_DRVVBUS
```

BigJacek, 2020-06-19:
ok.

BigJacek, 2020-06-19:
I was just about to write it.
;)

rsglobal, 2020-06-19:
@BigJacek , @vikitinho 
I've updated images, could you test it please?
https://drive.google.com/file/d/1AnSOOrhmPjvxW9sMMOE_NzwxXdhIR50t/view?usp=sharing

BigJacek, 2020-06-19:
Roman,
I'm starting to work on this.
JaS

BigJacek, 2020-06-19:
Brief summary:
I flashed Android on sdcard using a new deploy-sd.img.
Inserted recovery sdcard into the OPP2 board.
Connected microusb cable to OTG connector and my PC.
Powered-up the board.
After a short while the system (Windows 10) detected a new device in the Universal Serial Bus Controllers section (Unknown USB device (port reset failed).
 Since I'm working on Ubuntu using VM I can't remap this device on VMware (just can't see it). 
The difference is that the device is already in the system.
I tried also fastboot reboot-recovery in Windows but still  < waiting for any device > only and nothing more...

BigJacek, 2020-06-19:
maybe @vikitinho will have more luck

BigJacek, 2020-06-19:
I tried to install various drivers for this device in Windows but without success (even in unsigned driver installation mode).

vikitinho, 2020-06-19:
I'm starting to work on this.

I return with information in a few hours

vikitinho, 2020-06-19:
Boot OK:

[minicom-19062020.log](https://github.com/GloDroid/glodroid_manifest/files/4805481/minicom-19062020.log)

Fastboot nOK

I believe there is a problem with the driver on ubuntu, you will analyze and analyze information later.

ubuntu@ubuntu:~$ dmesg -w
[ 3055.245373] usb usb3-port1: Cannot enable. Maybe the USB cable is bad?
[ 3056.213471] usb usb3-port1: Cannot enable. Maybe the USB cable is bad?
[ 3056.213557] usb usb3-port1: attempt power cycle
[ 3057.497439] usb usb3-port1: Cannot enable. Maybe the USB cable is bad?
[ 3058.465445] usb usb3-port1: Cannot enable. Maybe the USB cable is bad?
[ 3058.465531] usb usb3-port1: unable to enumerate USB device

rsglobal, 2020-06-19:
@vikitinho @BigJacek ,

Thanks for testing. The problem with USB enumeration is from device side. Lets make 2 more attempts with different DTS node enabled.

rsglobal, 2020-06-19:
Attempt #3: https://drive.google.com/file/d/1w5fIzY07AYOdB8IrwH5xJ9G9npy6C9jE/view?usp=sharing

BigJacek, 2020-06-19:
ok. 
I'm starting to work on this.

BigJacek, 2020-06-19:
Brief summary:
Windows 10  detected a new device in the Universal Serial Bus Controllers section (Unknown USB device (port reset failed).
new:
at now it is possible to remap this device on VMware and is visible on ubuntu as "google usb download gadget".
I tried "fastboot reboot-recovery" in Windows and Ubuntu also but still < waiting for any device > only and nothing more...

BigJacek, 2020-06-19:
"fastboot device" doesn't recognize any device(s)

vikitinho, 2020-06-19:
Hi!

Fastboot not work.

[minicom-19062020-1.log](https://github.com/GloDroid/glodroid_manifest/files/4806240/minicom-19062020-1.log)

ubuntu@ubuntu:~$ dmesg
[13859.711875] usb 3-4: new high-speed USB device number 60 using xhci_hcd
[13859.860176] usb 3-4: New USB device found, idVendor=18d1, idProduct=4ee0
[13859.860180] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[13859.860183] usb 3-4: Product: USB download gadget
[13859.860186] usb 3-4: Manufacturer: Allwinner Technology
[13859.860189] usb 3-4: SerialNumber: 02c000819bd7bda4
ubuntu@ubuntu:~$ ./fastboot reboot
< waiting for any device >

rsglobal, 2020-06-19:
#4: https://drive.google.com/file/d/1DjiIPRoVMcGxxN0JPPt_Ly2Wo_xcxSko/view?usp=sharing

rsglobal, 2020-06-19:
@vikitinho , 

It seems that Number 3 works for you.
Try ```sudo ./fastboot reboot```

BigJacek, 2020-06-19:
@rsglobal 
It seems that Number 3 works similar to my as "google usb download gadget".
I tried ./fastboot reboot in sudo mode but < waiting for any device > only

BigJacek, 2020-06-19:
I'm starting to work on attempt 4

vikitinho, 2020-06-19:
```
ubuntu@ubuntu:~$ sudo ./fastboot reboot
Rebooting                                          OKAY [  0.000s]
Finished. Total time: 0.552s
```

I'll install and come back with more information

BigJacek, 2020-06-19:
ok 
probably there is a problem to remap this device on VMware -> however it is visible as "google usb download gadget" but not directly on ubuntu (VMware indicator light is probably cheating on something).

@vikitinho in you the only hope

BigJacek, 2020-06-19:
I'll keep track of it. Maybe something else will enlighten me.

[24 more replies]

closed 2020-06-23 by rsglobal