GloDroid/glodroid_manifest #17: Known bugs

hostkit, 2020-04-26:
Os: Your building img v0.2
Device: plus2e
Problem:
1. Leanback launcher and Android tv launcher Fc
2. Unable download using default browser, we must using fdroid first time to install another browser

Ask:
1. How to enable ir remote tv/set top box?
2. Must we installing SuperSU to enable root? Root Explorer not detect root

hostkit, 2020-05-10:
News bugs:
1. Android System: There's an internal problem with your device. Contact your manufacturer for details.
2. Usb storage/hhd external not mount and not showing in ls /dev, so unable to mount manual, i thinks internal mmc too

rsglobal, 2020-06-29:
> 2. Usb storage/hhd external not mount and not showing in ls /dev, so unable to mount manual, i thinks internal mmc too

It seems that there is no support for NTFS file system mount in AOSP yet. Corresponding change is under review.
https://android-review.googlesource.com/c/platform/system/vold/+/972633

rsglobal, 2020-09-23:
@hostkit , Please try to reproduce bugs with v0.5.0

rsglobal, 2020-10-01:
Closing due to inactivity. Please reopen if issue is still relevant.

closed 2020-10-01 by rsglobal