GloDroid/glodroid_manifest #19: Integrate and test MicroG package set

rsglobal, 2020-05-29:
GmsCore:
https://github.com/microg/android_packages_apps_GmsCore

memiks, 2020-07-17:
Hi @rsglobal , I would like to help you to add MicroG inside GloDroid, in order to have Free Android image for PinePhone.

Could you help me to understand how to modify GloDroid tools in order to add patch for signature spoofing during building of GloDroid ?

Thanks,
Memiks

rsglobal, 2020-07-17:
Hello @memiks ,

Thank you for your interest in GloDroid,

I would like to avoid any modification to AOSP part, that are required to integrate MicroG.
I rather see MicroG support as a separate patch archive, that user will put into GloDroid tree, run and then build the project..

At least for now.

Regards,
Roman

memiks, 2020-07-24:
Hi @rsglobal ,

As you explained, you would like to avoid modification of AOSP and I understand that completely.
That is why I managed to create a patch need to be applied just before make images command.

Please find it here --> [glodroid_microg_patch.diff.txt](https://github.com/GloDroid/glodroid_manifest/files/4969821/glodroid_microg_patch.diff.txt)

so basically:
`#copy patch file in glodroid repository`
`cd glodroid`
`repo sync -cq`

`source ./build/envsetup.sh`
`lunch pinephone-userdebug`
`#make clean`

`cd ./frameworks/base/`
`git apply ../../glodroid_microg_patch.diff`
`make images`

I hope it could help ;)

BR,
Memiks.

jaymeanchante, 2020-10-22:
Hey @memiks do you have a built image with microG? Can you share more information on how to do it for a beginner like me? I would like to use microG with GloDroid

herbsmn, 2022-11-04:
This issue looks like it can be closed since it seems to be outside of the scope of the GloDroid project.

closed 2022-11-04 by rsglobal