GloDroid/glodroid_manifest #20: Feature Request: Custom Recovery

hamiltonrn, 2020-06-12:
If possible, please add a custom recovery like TWRP.

rsglobal, 2020-06-12:
Hello,

Thank you for your request.

Our main mission, is to bring-up native mainline AOSP to the inexpensive and powerful SBCs.
And we have a lot of things to do ahead to achieve that goal.

Maybe in some distant future we will start to customize AOSP, but we haven't such plans yet.

Maybe you a looking for some particular feature from TWRP, that we can help enable with AOSP recovery?

Regards,
Roman

hamiltonrn, 2020-06-14:
> Maybe you a looking for some particular feature from TWRP, that we can help enable with AOSP recovery?
> 
> Regards,
> Roman

Usually I use the File Manager and Nandroid Backup. It's possible to flash Magisk with AOSP Recovery? I'm waiting my PinePhone to arrive, can't test it now.

Thank you for your reply.

NO-ob, 2020-07-07:
> Usually I use the File Manager and Nandroid Backup. It's possible to flash Magisk with AOSP Recovery? I'm waiting my PinePhone to arrive, can't test it now.
>
> Thank you for your reply.

Heres a magisk patched boot image it can be flashed with ```fastboot flash boot_a filename```

https://z.zz.ht/ZO46U.img

sonicpp, 2020-10-17:
@NO-ob : thanks for a tip! I tested it with GloDroid 0.5.1 (PinePhone), but when I tried `fastboot flash boot_a ZO46U.img`, it failed with `fastboot: error: boot partition is smaller than boot image`. So I tried it to restart with `fastboot reboot-fastboot`, where I could flash it:
```
Sending 'boot_a' (19016 KB)                        OKAY [  1.461s]
Writing 'boot_a'                                   OKAY [  2.293s]
```

But since then the phone won't boot up (I guess it overwrote data behind boot_a) - it is not big deal for me, since I had GloDroid for testing only, but...how did you manage to install it?

NO-ob, 2020-10-17:
It probably wont work because 0.5.1 is android 11 and that boot image is for android 10. I made it in the magisk manager it lets you patch boot image files.

jaymeanchante, 2020-10-22:
@NO-ob I installed GloDroid 0.5.1 to the eMMC to the PinePhone pmOS convergence edition, everything worked just fine. Then I patched the boot.img using Magisk Manager and flashed it with `fastboot flash boot_a magisk_patched.img` then the phone won't boot and stays in a bootloop, do you have any tips? Can you upload a new patched boot.img?

jaymeanchante, 2020-10-22:
@rsglobal I would use a custom recovery to flash a zip like Magisk in order to have root access and to be able to install microG, that would be awesome

herbsmn, 2022-11-04:
I found this on XDA...

[Guide] How to root GloDroid 12 on the PinePhone64 https://forum.xda-developers.com/t/guide-how-to-root-glodroid-12-on-the-pinephone64.4404449/

rsglobal, 2023-05-02:
We do not have any plans to ship custom recovery. But rooting using magic should be already possible.
Closing this for now. If you have more rationale on why to use a custom recovery, please reopen this issue and explain it.

closed 2023-05-02 by rsglobal

PublicVoidUpdate, 2023-06-23:
You can access fastboot and install a root like magisk on glodroid already. I used magisk to get a aurora store root install. its a bit of a process, but not much different than a normal root setup.