GloDroid/glodroid_manifest #21: Build failing due to missing file

Uclydde, 2020-06-18:
$ make sdcard
============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=10
TARGET_PRODUCT=pinephone
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a53
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=cortex-a15
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-5.7.1-050701-generic-x86_64-KDE-neon-User-Edition-5.19
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=QQ2A.200305.004.A1
OUT_DIR=out
============================================
[  0% 4/27791] Creating sdcard image...
FAILED: out/target/product/pinephone/sdcard.img
/bin/bash -c "PATH=/sbin:/bin:/usr/bin:\$PATH device/glodroid/platform/tools/gensdimg.sh -C=out/target/product/pinephone -T=SD -P=sunxi out/target/product/pinephone/sdcard.img"
===> Create raw disk image
dd: failed to open 'out/target/product/pinephone/sdcard.img': No such file or directory
[  0% 6/27791] Target dtb image: out/target/product/pinephone/boot_dtbo.img
FAILED: out/target/product/pinephone/boot_dtbo.img
/bin/bash -c "out/host/linux-x86/bin/mkdtboimg.py cfg_create out/target/product/pinephone/boot_dtbo.img out/target/product/pinephone/gen/DTBO/dtbo.cfg"
Traceback (most recent call last):
  File "/home/julian/GloDroid/out/host/linux-x86/bin/mkdtboimg.py/internal/stdlib/runpy.py", line 174, in _run_module_as_main
  File "/home/julian/GloDroid/out/host/linux-x86/bin/mkdtboimg.py/internal/stdlib/runpy.py", line 72, in _run_code
  File "/home/julian/GloDroid/out/host/linux-x86/bin/mkdtboimg.py/__main__.py", line 12, in <module>
  File "/home/julian/GloDroid/out/host/linux-x86/bin/mkdtboimg.py/internal/stdlib/runpy.py", line 174, in _run_module_as_main
  File "/home/julian/GloDroid/out/host/linux-x86/bin/mkdtboimg.py/internal/stdlib/runpy.py", line 72, in _run_code
  File "/home/julian/GloDroid/out/host/linux-x86/bin/mkdtboimg.py/mkdtboimg.py", line 1003, in <module>
  File "/home/julian/GloDroid/out/host/linux-x86/bin/mkdtboimg.py/mkdtboimg.py", line 1000, in main
  File "/home/julian/GloDroid/out/host/linux-x86/bin/mkdtboimg.py/mkdtboimg.py", line 857, in create_dtbo_image_from_config
IOError: [Errno 2] No such file or directory: '/home/julian/GloDroid/out/target/product/pinephone/obj/KERNEL_OBJ/arch/arm64/boot/dts/allwinner/sun50i-a64-pinephone.dtb'
22:48:30 ninja failed with: exit status 1

#### failed to build some targets (10 seconds) ####

rsglobal, 2020-06-18:
Hello,

Could you please share all commands was used t build starting from ```repo init```?

Uclydde, 2020-06-18:
Hi,
I ran all of the commands listed in the readme, and built for the pinephone:

`mkdir -p GloDroid`
`cd GloDroid`
`repo init -u https://github.com/glodroid/glodroid_manifest`
`repo sync -cq`
`source ./build/envsetup.sh`
`lunch`
`pinephone-userdebug`
`make sdcard`

Would you like the output of any of them?

rsglobal, 2020-06-18:
Thanks for reporting this issue.
You are building master branch, which is currently under development.
Instead you can try latest pre-release:
```repo init -u https://github.com/glodroid/glodroid_manifest -b refs/tags/v0.2.0```

Uclydde, 2020-06-18:
The build still fails. Here's the output:

[  9% 80/815] //frameworks/base/packages/SystemUI:SystemUI-core kotlinc [common]
frameworks/base/packages/SystemUI/src/com/android/systemui/statusbar/PulseExpansionHandler.kt:121:38: warning: 'get(Class<T!>!): T!' is deprecated. Deprecated in Java
        mFalsingManager = Dependency.get(FalsingManager::class.java)
                                     ^
[  9% 81/815] //frameworks/base:system-api-stubs-docs Metalava Check API [common]
FAILED: out/soong/.intermediates/frameworks/base/system-api-stubs-docs/android_common/check_last_released_api.timestamp out/soong/.intermediates/frameworks/base/system-api-stubs-docs/android_common/last_released_baseline.txt
( rm -rf "out/soong/.intermediates/frameworks/base/system-api-stubs-docs/android_common/last-apicheck/srcjars" && mkdir -p "out/soong/.intermediates/frameworks/base/system-api-stubs-docs/android_common/last-apicheck/srcjars" && out/soong/host/linux-x86/bin/zipsync -d out/soong/.intermediates/frameworks/base/system-api-stubs-docs/android_common/last-apicheck/srcjars -l out/soong/.intermediates/frameworks/base/system-api-stubs-docs/android_common/last-apicheck/srcjars/list -f "*.java" out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/privacy.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/section.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/activitymanager.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/alarmmanager.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/enums.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/notification.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/notification_channel.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/notification_channel_group.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/notificationmanager.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/pendingintent.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/profilerinfo.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/settings_enums.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/statusbarmanager.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/window_configuration.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/app/job/enums.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/bluetooth/enums.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/bluetooth/a2dp/enums.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/bluetooth/hci/enums.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/bluetooth/hfp/enums.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/bluetooth/smp/enums.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/content/activityinfo.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/content/clipdata.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/content/clipdescription.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/content/component_name.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/content/configuration.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/content/featureinfo.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/content/intent.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/content/locale.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/content/package_item_info.srcjar out/soong/.intermediates/frameworks/base/framework-javastream-protos/gen/gensrcs/frameworks/base/core/proto/android/debug/enums.srcjar out/soong/.intermediates/frameworks/base/framework [... cut ...] ho -e "\n******************************\nYou have tried to change the API from what has been previously released in\nan SDK.  Please fix the errors listed above.\n******************************\n" ; exit 38 )
Killed
-e 
******************************
You have tried to change the API from what has been previously released in
an SDK.  Please fix the errors listed above.
******************************

15:36:19 ninja failed with: exit status 1

#### failed to build some targets (06:37 (mm:ss)) ####

rsglobal, 2020-06-18:
'Killed' usually means you are out of RAM.

Try to close unused applications. Also try build in 1 thread. ( `make -j1 sdcard` )

Uclydde, 2020-06-22:
Thank you for your help. Building on one thread fixed that problem. Though I did have to install mtools, because mcopy was missing from my system.

closed 2020-06-22 by Uclydde