GloDroid/glodroid_manifest #23: [Request] Android 10 port for Orange Pi Prime / PC2

Quang-vip, 2020-06-23:
I'm wondering if we could port Android 10 to Allwinner H5 devices? The H5 has been supported by mainline kernel since 4.14 (or 4.18, I'm not sure). 

And most of the Prime's drivers have also been integrated into the kernel since then. So I guess we can make a port for it?

rsglobal, 2020-06-23:
I do not have this boards, but if you can help with testing we could try/

Quang-vip, 2020-06-23:
Alright, I'll be more than happy to help you with that ;)
(Android 5.1 is buggy af to be honest)

rsglobal, 2020-06-23:
opi-prime v1:
https://drive.google.com/file/d/1xCNAPStrLd3JXrRGTnQ950IMriKbRmom/view?usp=sharing
Use https://github.com/GloDroid/glodroid_manifest/blob/next/README.md as deploying manual.

Quang-vip, 2020-06-24:
Hmmm, that's weird... It almost worked

At first, my PC recognized it as "USB Download Gadget" then "LeMobile Android Bootloader"
But then it said the USB device was recognized...

Probably because of my 8GB Class 4 SD Card, well, lemme try the 32GB Class 10 one

rsglobal, 2020-06-24:
It would be very helpful if you can attach ttl-uart usb adapter and send me logs.

Quang-vip, 2020-06-24:
[image]

Alright, I got'em. I managed to get it to boot using the 8GB one (my PC recognized it as LeMobile Android device, weird.. but it worked :D)

Now the issues...
I guess for stability and speed, I'll try again with the 32GB one. And for now:
+ The mouse movement is kinda stuttering
+ I can't pull the status bar down
+ I could hardly launch any apps on the home screen (even if either of them launched, it just hung)
+ The navigation bar buttons are clickable, but their functions are kinda unusable
+ It seems like it's using software rendering..

rsglobal, 2020-06-24:
@Quang-vip ,

Thank you for testing.
It seems that you are using 1080p screen, and Android lack contiguous memory for graphic buffers.
It uses LIMA GPU driver for rendering, but unfortunately without GPU offloading by composer hardware for now.

Can you open settings and check if wifi works?

Quang-vip, 2020-06-24:
@rsglobal 
Well, I tried to pull the apps list up but then it hung
Lemme try again

rsglobal, 2020-06-24:
I've increased video memory from 128M to 256M, 
UI should be more stable.

V2: https://drive.google.com/file/d/1MDY6srLyZZ2v_4BagVw-09ddaNfQQKcG/view?usp=sharing

rsglobal, 2020-06-24:
Please also check audio functionality

Quang-vip, 2020-06-24:
@rsglobal
Is there any way to enter fastboot mode without having to send the adb command to it?

The settings app is really messed up, clicking on sub-menus doesn't open it up

Quang-vip, 2020-06-24:
[image]

Alright, we've got more positive results here

Wifi and bluetooth don't work but ethernet does, well, except its LEDs keep flashing simultaneously. But still, Internet is working fine through ethernet.

Audio is also working thru the 3.5mm jack, I haven't tested HDMI audio but I think it also works.

And yea, the UI is much more stable now ;))

ADB is also working after I re-enable USB debugging through developer options.

rsglobal, 2020-06-24:
> [Edit] After a few minutes of video playing, it suddenly froze

I may guess SOC is overheated. Throttling may not enabled for now. Try to install heatsink.

Could you send me output of:
```
adb shell dmesg
adb shell logcat
adb shell lsmod
```

Thanks in advance.

Quang-vip, 2020-06-24:
Alright, here's:
+ dmesg: https://pastebin.com/2zBD9r04
+ logcat: [logcat.txt](https://github.com/GloDroid/glodroid_manifest/files/4825280/logcat.txt)
+ lsmod: [lsmod.txt](https://github.com/GloDroid/glodroid_manifest/files/4825281/lsmod.txt)

rsglobal, 2020-06-24:
Thanks.
Please check if wifi works on v3: https://drive.google.com/file/d/1QIgEryuyQq2QhedWG2YhWVPEccMM_tqp/view?usp=sharing

Quang-vip, 2020-06-24:
@rsglobal 
Wifi is confirmed to be working on the v3 image ;)

rsglobal, 2020-06-24:
PC2: https://drive.google.com/file/d/11-MIB1D7Q0WJ0Ocpkgswyu6RKM06LI_2/view?usp=sharing

rsglobal, 2020-06-24:
@Quang-vip , would you have a chance to test opi-pc2 in a nearest time?

Quang-vip, 2020-06-24:
@rsglobal
Sadly, I only own an OPi Prime :((

But I think we could put that on the Orange pi forum and ask if anyone wants to help?

rsglobal, 2020-06-25:
I see.

Initial bring-up of opi-prime finished, available in next branch and will be included into next release.

closed 2020-06-25 by rsglobal

fiamko, 2021-07-27:
I have an OPI PC2 and I would like to help, but the links are no longer accessible ...

rsglobal, 2021-07-27:
Link was for test image that didn't worked. You may try downloading/building PC2 from the sources.

rsglobal, 2021-07-27:
Latest image can be downloaded here: https://github.com/GloDroid/glodroid_manifest/releases/tag/v0.6.1

RamagaGrodno, 2022-11-20:
hello, AND THERE IS A 10 ANDROID FIRMWARE IN .img FORMAT so that you can flash it or run it from a USB flash drive ?. phonix card software? Thanks.