GloDroid/glodroid_manifest #27: Not display on PinePhone

osakanataro, 2020-07-04:
I tried to following step on PinePhone.
1. get gd-pinephone-v0.3.0.tar.gz and extract
2. create boot microSD , "sudo bmaptool copy deploy-sd.img /dev/sdb --nobmap"
3. insert microSD
4. connect PinePhone and Ubuntu PC
5. PinePhone into fastboot mode ,but no display.
6. execute flash-sd.sh on Ubuntu PC. start to write on PinePhone
7. PinePhone reboot, but no display...
8. execute "adb logcat" on Ubuntu PC

PinePhone was no display and no LED.
But, adb shell is worked.

`07-01 13:16:21.618   342   395 E hwc-platform-drm-generic: failed to import prime fd 20 ret=-1
07-01 13:16:21.618   342   395 E hwc-drm-two: Failed to import layer, ret=-1
07-01 13:16:21.619   342   395 E hwc-platform-drm-generic: failed to import prime fd 19 ret=-1
07-01 13:16:21.620   342   395 E hwc-drm-two: Failed to import layer, ret=-1
07-01 13:16:21.620   354   354 E HWComposer: presentAndGetReleaseFences: present failed for display 0: NoResources (6)
07-01 13:16:21.645   342   395 E hwc-platform-drm-generic: failed to import prime fd 19 ret=-1
07-01 13:16:21.645   342   395 E hwc-drm-two: Failed to import layer, ret=-1
07-01 13:16:21.648   342   395 E hwc-platform-drm-generic: failed to import prime fd 20 ret=-1
07-01 13:16:21.649   342   395 E hwc-drm-two: Failed to import layer, ret=-1
07-01 13:16:21.649   354   354 E HWComposer: presentAndGetReleaseFences: present failed for display 0: NoResources (6)
07-01 13:16:21.673   342   395 E hwc-platform-drm-generic: failed to import prime fd 20 ret=-1
07-01 13:16:21.673   342   395 E hwc-drm-two: Failed to import layer, ret=-1
07-01 13:16:21.676   342   395 E hwc-platform-drm-generic: failed to import prime fd 19 ret=-1
07-01 13:16:21.677   342   395 E hwc-drm-two: Failed to import layer, ret=-1
07-01 13:16:21.677   354   354 E HWComposer: presentAndGetReleaseFences: present failed for display 0: NoResources (6)
07-01 13:16:21.699   342   395 E hwc-platform-drm-generic: failed to import prime fd 19 ret=-1
07-01 13:16:21.699   342   395 E hwc-drm-two: Failed to import layer, ret=-1
07-01 13:16:21.705   342   395 E hwc-platform-drm-generic: failed to import prime fd 20 ret=-1
07-01 13:16:21.705   342   395 E hwc-drm-two: Failed to import layer, ret=-1
07-01 13:16:21.706   354   354 E HWComposer: presentAndGetReleaseFences: present failed for display 0: NoResources (6)
07-01 13:16:21.729   342   395 E hwc-platform-drm-generic: failed to import prime fd 20 ret=-1
07-01 13:16:21.729   342   395 E hwc-drm-two: Failed to import layer, ret=-1
07-01 13:16:21.731   342   395 E hwc-platform-drm-generic: failed to import prime fd 19 ret=-1
07-01 13:16:21.731   342   395 E hwc-drm-two: Failed to import layer, ret=-1
07-01 13:16:21.731   354   354 E HWComposer: presentAndGetReleaseFences: present failed for display 0: NoResources (6)
07-01 13:16:21.757   342   395 E hwc-platform-drm-generic: failed to import prime fd 19 ret=-1
07-01 13:16:21.758   342   395 E hwc-drm-two: Failed to import layer, ret=-1
07-01 13:16:21.761   342   395 E hwc-platform-drm-generic: failed to import prime fd 20 ret=-1
07-01 13:16:21.767   342   395 E hwc-drm-two: Failed to import layer, ret=-1
07-01 13:16:21.767   354   354 E HWComposer: presentAndGetReleaseFences: present failed for display 0: NoResources (6)
07-01 13:16:21.785   342   395 E hwc-platform-drm-generic: failed to import prime fd 20 ret=-1
07-01 13:16:21.785   342   395 E hwc-drm-two: Failed to import layer, ret=-1
07-01 13:16:21.788   342   395 E hwc-platform-drm-generic: failed to import prime fd 19 ret=-1
07-01 13:16:21.788   342   395 E hwc-drm-two: Failed to import layer, ret=-1
07-01 13:16:21.789   354   354 E HWComposer: presentAndGetReleaseFences: present failed for display 0: NoResources (6)`

HomebrewXS, 2020-07-04:
I have the same problem. Android is booting but you can’t see anything. You can touch things and you hear the “touch” sound. But it doesn’t show anything :/

rsglobal, 2020-07-04:
@osakanataro ,

Thank you for testing.
Can you provide detailed logs?
```bash
adb shell logcat | tee logcat.txt
adb shell dmesg | tee dmesg.txt
adb shell lsmod | tee lsmod.txt
adb shell lshal | tee lshal.txt
```
And attach files to the issue.

ch0rus, 2020-07-04:
I'm trying to get GloDroid running on Pine64+ and have exactly the same problem with drm

The HDMI framebuffer is ok, data sent to fb0 is displayed

rsglobal, 2020-07-04:
On v0.2.0 /dev/dri/card0 and /dev/dri/card1 was swapped for some reason comparing to other products, therefore Icenowy Zheng had to change device that used by hwc to card1.
Maybe on v0.3.0 drm and gpu order changed once again.

ch0rus, 2020-07-04:
I have tried both card0 and card1, no difference. But maybe my problem is something else.

I will collect some logs later today

rsglobal, 2020-07-04:
According to the logs provided by @osakanataro , it's very likely that hwc try to use GPU as KMSRO.
I just want to make sure you've did all correctly.
You have to modify:
https://github.com/GloDroid/glodroid_device/blob/master/pinephone/drm.pinephone.rc
Both hwc and gralloc fields to card0.
At runtime make sure properties are applied before and after your modifications:
`adb shell getprop | grep card`

osakanataro, 2020-07-05:
> @osakanataro ,
> 
> Thank you for testing.
> Can you provide detailed logs?
> 
> ```shell
> adb shell logcat | tee logcat.txt
> adb shell dmesg | tee dmesg.txt
> adb shell lsmod | tee lsmod.txt
> adb shell lshal | tee lshal.txt
> ```
> 
> And attach files to the issue.
> 
> ADD: Please attach also output of `adb shell getprop | tee getprop.txt`

[getprop.txt](https://github.com/GloDroid/glodroid_manifest/files/4874367/getprop.txt)
[logcat.txt](https://github.com/GloDroid/glodroid_manifest/files/4874368/logcat.txt)
[lshal.txt](https://github.com/GloDroid/glodroid_manifest/files/4874369/lshal.txt)
[lsmod.txt](https://github.com/GloDroid/glodroid_manifest/files/4874370/lsmod.txt)
[dmesg.txt](https://github.com/GloDroid/glodroid_manifest/files/4874371/dmesg.txt)

rsglobal, 2020-07-05:
@osakanataro ,
Thank you.
Please also share `adb pull /proc/config.gz`

Does v0.2.0 works fine for you? If so, please share same data for comparison.

osakanataro, 2020-07-05:
v0.3.0 config.gz
[config.gz](https://github.com/GloDroid/glodroid_manifest/files/4874666/config.gz)

v0.2.0 worked fine.
[v020-config.gz](https://github.com/GloDroid/glodroid_manifest/files/4874674/v020-config.gz)
[v020-dmesg.txt](https://github.com/GloDroid/glodroid_manifest/files/4874675/v020-dmesg.txt)
[v020-getprop.txt](https://github.com/GloDroid/glodroid_manifest/files/4874676/v020-getprop.txt)
[v020-logcat.txt](https://github.com/GloDroid/glodroid_manifest/files/4874677/v020-logcat.txt)
[v020-lshal.txt](https://github.com/GloDroid/glodroid_manifest/files/4874678/v020-lshal.txt)
[v020-lsmod.txt](https://github.com/GloDroid/glodroid_manifest/files/4874679/v020-lsmod.txt)

rsglobal, 2020-07-05:
Thanks.

Please try to replace boot.img and flash again.

[boot.zip](https://github.com/GloDroid/glodroid_manifest/files/4874798/boot.zip)

osakanataro, 2020-07-05:
I tried replace boot.img, but not display. 
"Android logo" not display.
[v030a-getprop.txt](https://github.com/GloDroid/glodroid_manifest/files/4874848/v030a-getprop.txt)
[v030a-logcat.txt](https://github.com/GloDroid/glodroid_manifest/files/4874849/v030a-logcat.txt)
[v030a-lshal.txt](https://github.com/GloDroid/glodroid_manifest/files/4874850/v030a-lshal.txt)
[v030a-lsmod.txt](https://github.com/GloDroid/glodroid_manifest/files/4874851/v030a-lsmod.txt)
[v030a-config.gz](https://github.com/GloDroid/glodroid_manifest/files/4874852/v030a-config.gz)
[v030a-dmesg.txt](https://github.com/GloDroid/glodroid_manifest/files/4874853/v030a-dmesg.txt)

ch0rus, 2020-07-05:
@rsglobal I opened a separate issue for Pine64
#29

rsglobal, 2020-07-05:
@osakanataro ,

Please test this image.
https://drive.google.com/file/d/1Z7Ajtk2Hws_PIVFtoU6JzkMsd__rJsSe/view?usp=sharing
No need to gather all logs this time.
Just in case UI won't work, please attach output of 
```
adb root
adb shell lsof | tee lsof.txt
```

osakanataro, 2020-07-06:
pinephone-r2.tar is worked fine.
https://photos.app.goo.gl/6HLzduRxfr7n5zfq8

[v030-r2-lsof.txt](https://github.com/GloDroid/glodroid_manifest/files/4878315/v030-r2-lsof.txt)

rsglobal, 2020-07-06:
Thanks.
Could you also check r3?
https://drive.google.com/file/d/1ZvvhipgHV7X2wRHH9N-obd1m_M5n3mqa/view?usp=sharing

osakanataro, 2020-07-06:
pinephone-r3.tar is worked too.

[v030-r3-lsof.txt.gz](https://github.com/GloDroid/glodroid_manifest/files/4878475/v030-r3-lsof.txt.gz)

rsglobal, 2020-07-06:
That;s great.
Does booting from eMMC work?

osakanataro, 2020-07-06:
Sorry, I use eMMC for otherway. 
I could not write to eMMC.

rsglobal, 2020-07-06:
v0.3.0 updated with the fix included.
Assets was also updated.

closed 2020-07-06 by rsglobal