GloDroid/glodroid_manifest #28: Installation failure on orangepi plus2e

hadimz, 2020-07-04:
When trying to power up Orangepi plue2e using sd card with the deploy-sd-for-emmc.img file, nothing shows up on the monitor and the flash-emmc.sh command ends in the following error.
+ ./fastboot flash gpt deploy-gpt.img
./flash-emmc.sh: line 4: ./fastboot: cannot execute binary file

rsglobal, 2020-07-04:
Hello

Is your host pc mac or x86 linux?

hadimz, 2020-07-04:
Hi,
It's mac

rsglobal, 2020-07-04:
Fastboot,adb and mk2fs binaries was compiled for linux only.
You have to replace them with corresponding tools that built for mac.
https://developer.android.com/studio/releases/platform-tools

hadimz, 2020-07-04:
Thank you, I'll try it out.
isn't the board supposed to boot from sd? its leds don't light up and nothing shows up on the monitor. is this expected?

hadimz, 2020-07-04:
Thank you. It worked like a charm. I installed android platform tools and removed the './' from fastboot commands in the 'flash-emmc.sh' file and it resolved the issue.

closed 2020-07-04 by hadimz