GloDroid/glodroid_manifest #29: Request for help with porting to Pine64

ch0rus, 2020-07-05:
I guess it's best to have a separate issue for Pine64

Props are set like this:
    setprop hwc.drm.device /dev/dri/card1
    setprop gralloc.gbm.device /dev/dri/card1

I verified that card1 the correct device

Logs
[lsof2.txt](https://github.com/GloDroid/glodroid_manifest/files/4875452/lsof2.txt)
[lshal2.txt](https://github.com/GloDroid/glodroid_manifest/files/4875453/lshal2.txt)
[logcat2.txt](https://github.com/GloDroid/glodroid_manifest/files/4875454/logcat2.txt)
[dmesg2.txt](https://github.com/GloDroid/glodroid_manifest/files/4875455/dmesg2.txt)
[dmesg2-drm-debug.txt](https://github.com/GloDroid/glodroid_manifest/files/4875456/dmesg2-drm-debug.txt)

Not sure why allocator uses card0

If I set hwc.drm.use_overlay_planes=0 I managed to get boot logo showing

rsglobal, 2020-07-05:
Here is opi3 lsof:
```
opi3:/ # lsof | grep card                                                                                                                                                                                
allocator@2.0-s   346     system    5u      CHR              226,0       0t0      11750 /dev/dri/card0
allocator@2.0-s   346     system    6u      CHR              226,0       0t0      11750 /dev/dri/card0
composer@2.1-se   347     system    6u      CHR              226,0       0t0      11750 /dev/dri/card0
composer@2.1-se   347     system   12u      CHR              226,0       0t0      11750 /dev/dri/card0
composer@2.1-se   347     system   13u      CHR              226,0       0t0      11750 /dev/dri/card0
surfaceflinger   359     system   19u      CHR              226,0       0t0      11750 /dev/dri/card0
surfaceflinger   359     system   20u      CHR              226,0       0t0      11750 /dev/dri/card0
ndroid.systemui   605     u0_a77  103u      CHR              226,0       0t0      11750 /dev/dri/card0
ndroid.systemui   605     u0_a77  104u      CHR              226,0       0t0      11750 /dev/dri/card0
ndroid.settings   724     system   48u      CHR              226,0       0t0      11750 /dev/dri/card0
ndroid.settings   724     system   49u      CHR              226,0       0t0      11750 /dev/dri/card0
droid.launcher3   829     u0_a75   70u      CHR              226,0       0t0      11750 /dev/dri/card0
droid.launcher3   829     u0_a75   71u      CHR              226,0       0t0      11750 /dev/dri/card0
opi3:/ # lsof | grep card                                                                                                                                                                                
allocator@2.0-s   346     system    5u      CHR              226,0       0t0      11750 /dev/dri/card0
allocator@2.0-s   346     system    6u      CHR              226,0       0t0      11750 /dev/dri/card0
composer@2.1-se   347     system    6u      CHR              226,0       0t0      11750 /dev/dri/card0
composer@2.1-se   347     system   12u      CHR              226,0       0t0      11750 /dev/dri/card0
composer@2.1-se   347     system   13u      CHR              226,0       0t0      11750 /dev/dri/card0
surfaceflinger   359     system   19u      CHR              226,0       0t0      11750 /dev/dri/card0
surfaceflinger   359     system   20u      CHR              226,0       0t0      11750 /dev/dri/card0
ndroid.systemui   605     u0_a77  103u      CHR              226,0       0t0      11750 /dev/dri/card0
ndroid.systemui   605     u0_a77  104u      CHR              226,0       0t0      11750 /dev/dri/card0
ndroid.settings   724     system   48u      CHR              226,0       0t0      11750 /dev/dri/card0
ndroid.settings   724     system   49u      CHR              226,0       0t0      11750 /dev/dri/card0
droid.launcher3   829     u0_a75   70u      CHR              226,0       0t0      11750 /dev/dri/card0
droid.launcher3   829     u0_a75   71u      CHR              226,0       0t0      11750 /dev/dri/card0
```

I suspect that after updating from aosp10-r32 to aosp10-r39 allocator HAL begin to start earlier, and catch `setprop gralloc.gbm.device /dev/dri/card0` value from common/drm.rc
Please try to change those values to card1 also and check if it helps..

rsglobal, 2020-07-06:
I will compile GPU driver as a module, thus making order the same on all platforms.

rsglobal, 2020-07-06:
This commit fixed pinephone issue and should also fix your issue.
https://github.com/GloDroid/glodroid_device/commit/498c58c4b43a957799fb3db3bfaf77715cd7375f

ch0rus, 2020-07-06:
Thanks Roman, finally it's working!

Now I'm struggling to get the rtl8723bs driver working. I get a wlan0 interface, but I can not set status to UP for some reason. Not sure if some firmware needs to be loaded or if that's only for Bluetooth.

Also Android is very slow and laggy compared to my RK3066 board, is that only for me or a known problem?

ch0rus, 2020-07-06:
I'm also having some problems with the U-Boot bootscript.

This part fails (invalid magic) and corrupts fdt:
adtimg get dt --index=1 dtb_start dtb_size &&
 cp.b \$dtb_start \$dtboaddr \$dtb_size &&
 fdt resize 8192 &&
 fdt apply \$dtboaddr &&

I can't really understand what's supposed to be in index 1, seems like everything is in index 0

rsglobal, 2020-07-06:
Try to grep dmesg:
`adb shell dmesg | grep -i firmware`

Index 1 is empty.dts:
platform/kernel/kernel.mk:52:ANDROID_DTS_OVERLAY	?= $(LOCAL_PATH)/empty.dts
Slow UI is known problem. We need to fix sun4i-drm kernel driver and enable composer to make UI faster.
I can help more if I can see your work. Consider making pull-request for review.

ch0rus, 2020-07-06:
Nothing in dmesg regarding firmware

I will clean up all my changes and try to make a pull request, right now it's very ugly

Thanks a lot for your work with GloDroid!

ch0rus, 2020-07-06:
What's needed to get sun4i-drm running?

rsglobal, 2020-07-07:
https://github.com/GloDroid/glodroid_device/blob/5e441440397e5dcde7516ba1ef51ac30e08daa1d/common/device-common-sunxi.mk#L12

Set it to 0 or remove this line.

rsglobal, 2020-07-07:
As for fixing kernel driver issues I had several attempts, but it requires much more time to investigate/test /fix those issues.
https://lore.kernel.org/patchwork/project/lkml/list/?series=&submitter=24380&state=*&q=drm%2Fsun4i&archive=both&delegate=

ch0rus, 2020-07-07:
I wish I could help, but that's way too complicated for me

But graphics is a lot faster in Armbian?

rsglobal, 2020-07-08:
Armbian was working great even without hardware acceleration for some time.
IMHO it because it was designed to work in such condition.
Android UI should inspire user and requires support of various hardware acceleration to do so. 

Currently there are 2 ways for making Android UI faster:
1. Fix sun4i-drm kernel driver
2. Enable tiled buffers for all buffers except HWFB.

Second is much easier to make, but it should give performance only in theory. I can't tell for sure now if it does.

rsglobal, 2020-07-16:
Any progress with Pine64?

rsglobal, 2020-07-23:
@ch0rus ,

It turns out that tiled buffers + reducing wallpaper resolution significantly increases UI performance.

rsglobal, 2020-09-03:
Closing due to inactivity. Please reopen if it is still relevant.

closed 2020-09-03 by rsglobal