GloDroid/glodroid_manifest #30: Auto rotate inverted

nas00, 2020-07-06:
On the latest pinephone release, the screen auto rotates apps to the opposite direction of the physical orientation.

rsglobal, 2020-08-07:
It seems that X and Y coordinates of accelerometer are swapped. 
https://developer.android.com/guide/topics/sensors/sensors_overview#sensors-coords

rsglobal, 2020-08-07:
I've added correct mount matrix into pinephone.dtsi that fixed this issue for me.
```
		mount-matrix = "0", "-1", "0",
			       "1", "0", "0",
			       "0", "0", "1";
```

rsglobal, 2020-08-07:
Change is available in the next integration build:
https://github.com/GloDroid/glodroid_manifest/issues/58#issuecomment-670354906

closed 2020-08-07 by rsglobal

rsglobal, 2020-09-11:
IOS orientation: https://developer.apple.com/documentation/uikit/uiacceleration?language=objc