GloDroid/glodroid_manifest #31: Add feature to gbm_gralloc and drm_hwcomposer HALs that will search for KMS DRI card

rsglobal, 2020-07-07:
We can not always guarantee that KMS card will always have /dev/dri/card0 path, 
This path can depend on different factors,  e.g. DTS node order, driver compiled as a module or built-in, etc.
Both drm_hwcomposer and gbm_gralloc have to always use KMS driver, not GPU.

rsglobal, 2020-07-16:
@matviizorin ,

I've juts found same algorithm in libkmsxx:
https://github.com/tomba/kmsxx/blob/d9d5e44526a3b8065e4ae913032020d0bfd93103/kms%2B%2B/src/card.cpp#L42

matviizorin, 2020-07-16:
@rsglobal,

Thanks a lot, I'll try to implement it in the nearest time

matviizorin, 2020-07-17:
I've just created the MR into drm_hwcomposer ([MR#108](https://gitlab.freedesktop.org/drm-hwcomposer/drm-hwcomposer/-/merge_requests/108)).

matviizorin, 2020-07-22:
WIP branch which will be upstream to [gbm_gralloc](https://github.com/robherring/gbm_gralloc) - [kms-device](https://github.com/matviizorin/gbm_gralloc/tree/kms-device).

matviizorin, 2020-07-23:
PR to [gbm_gralloc](https://github.com/robherring/gbm_gralloc) has been created ([PR#16](https://github.com/robherring/gbm_gralloc/pull/16)).

matviizorin, 2020-07-26:
The [MR#108](https://gitlab.freedesktop.org/drm-hwcomposer/drm-hwcomposer/-/merge_requests/108) was merged.

rsglobal, 2020-07-28:
I can see this log on opi3:
```
01-01 01:13:44.994   349   401 E GRALLOC-GBM: Create BO!!!, size=1024x600, fmt=1, usage=5, modifier=0
07-28 20:49:42.377   349   401 E GRALLOC-GBM: Create BO!!!, size=1024x600, fmt=1, usage=4, modifier=0
07-28 20:49:42.398   349   401 E GRALLOC-GBM: Create BO!!!, size=1024x600, fmt=1, usage=4, modifier=0
07-28 20:49:42.410   349   349 E GRALLOC-GBM: Create BO!!!, size=1024x600, fmt=1, usage=4, modifier=0
07-28 20:49:42.416   349   401 E GRALLOC-GBM: Create BO!!!, size=1024x600, fmt=1, usage=4, modifier=0
07-28 20:49:42.535   349   401 E GRALLOC-GBM: Create BO!!!, size=1024x48, fmt=1, usage=4, modifier=0
07-28 20:49:42.553   923  1140 E GRALLOC-GBM: failed to open /dev/dri/card*
07-28 20:49:42.564   349   401 E GRALLOC-GBM: Create BO!!!, size=1024x48, fmt=1, usage=4, modifier=0
07-28 20:49:42.564   349   349 E GRALLOC-GBM: Create BO!!!, size=1x1, fmt=1, usage=4, modifier=0
07-28 20:49:42.566   923  1140 E GRALLOC-GBM: failed to open /dev/dri/card*
07-28 20:49:42.569   349  1240 E GRALLOC-GBM: Create BO!!!, size=1024x48, fmt=1, usage=4, modifier=0
07-28 20:49:42.579   349  1240 E GRALLOC-GBM: Create BO!!!, size=1024x48, fmt=1, usage=4, modifier=0
07-28 20:49:44.919   349   401 E GRALLOC-GBM: Create BO!!!, size=1024x600, fmt=1, usage=4, modifier=0
07-28 20:49:44.922   349   401 E GRALLOC-GBM: Create BO!!!, size=1024x600, fmt=1, usage=4, modifier=0
07-28 20:49:44.931   349  1240 E GRALLOC-GBM: Create BO!!!, size=1024x600, fmt=1, usage=4, modifier=0
07-28 20:49:44.936   349  1240 E GRALLOC-GBM: Create BO!!!, size=1024x600, fmt=1, usage=4, modifier=0
07-28 20:49:46.040   349  1240 E GRALLOC-GBM: Create BO!!!, size=1024x48, fmt=1, usage=4, modifier=0
07-28 20:49:46.041   349  1240 E GRALLOC-GBM: Create BO!!!, size=1024x48, fmt=1, usage=4, modifier=0
07-28 20:49:46.043   349  1240 E GRALLOC-GBM: Create BO!!!, size=1024x48, fmt=1, usage=4, modifier=0
07-28 20:49:46.045   349  1240 E GRALLOC-GBM: Create BO!!!, size=1024x48, fmt=1, usage=4, modifier=0
07-28 20:49:46.046  1308  1345 E GRALLOC-GBM: failed to open /dev/dri/card*
07-28 20:49:46.050   349  1240 E GRALLOC-GBM: Create BO!!!, size=1x1, fmt=1, usage=4, modifier=0
07-28 20:49:46.050  1308  1345 E GRALLOC-GBM: failed to open /dev/dri/card*
07-28 20:49:47.865   349  1240 E GRALLOC-GBM: Create BO!!!, size=1024x600, fmt=1, usage=4, modifier=0
```

rsglobal, 2020-08-03:
Feature was added into integration branch and successfully tested.
@matviizorin , Thank you for contribution.

closed 2020-08-03 by rsglobal