GloDroid/glodroid_manifest #32: Pinephone emmc install

dewitpj, 2020-07-08:
Hi,

I noted that you "needed" someone to test the eMMC install on PinePhone. I have done this - at the moment Wifi doesn't seem stable but let me know if you want any more tests.

Thanks !

rsglobal, 2020-07-08:
Hello,

Thank you for testing and confirmation that deploying to eMMC works.
Could you rename this issue to `Pinephone wifi issue` or open another issue and add more details?

dewitpj, 2020-07-08:
I'll leave this one open for the eMMC "thing" - I'll open another issue for the Wifi issue.

dewitpj, 2020-07-08:
Also - for future ref, here is how I did it:

I installed JumpDrive on a SDCard and booted the PinePhone.
Using DD I wrote deploy-gpt.img to the eMMC (was /dev/disk2 for me on a Mac)
I then wrote bootloader-emmc.img to /dev/disk2s1
I then wrote env.img to /dev/disk2s2

I then rebooted the device and ran flash-emmc.sh from your "r3" release for the display test :)

This flashed/rebooted (a few times) the device and I was greeted with Andriod booting - nice !

rsglobal, 2020-07-08:
Did you had a problem with classic procedure, using deploy-sd-for-emmc.img (described in README.md)?

bigc0127, 2020-07-08:
when I use deploy-sd-for-emmc.img (described in README.md) it doesn't boot at all...

dewitpj, 2020-07-08:
I didn't see that to be honest - I can reflash the phone to stock and try it again ?

rsglobal, 2020-07-08:
> I didn't see that to be honest - I can reflash the phone to stock and try it again ?

You can surely do that.

bigc0127, 2020-07-08:
does not boot at all when using deploy-sd-for-emmc.img 

the only way I have gotten it to boot is to flash r2 then flash r3 over it manually...

dewitpj, 2020-07-09:
@bigc0127 - Did you try run the flash after a few seconds ?

On my Pinephone the deploy-sd-for-emmc.img showed a black screen but the console port showed it was alive - after a reboot - and a removal of the SD Card, fastbootd booted.

So, IMHO, does it work - yes - can it be improved, yes - in all fairness, I don't know if there was a display update on the r2 build.

dewitpj, 2020-07-09:
Right - I think I replicated your find:

% ./flash-emmc.sh
+ ./fastboot flash gpt deploy-gpt.img
Sending 'gpt' (128 KB)                             OKAY [  0.016s]
Writing 'gpt'                                      OKAY [  0.015s]
Finished. Total time: 0.046s
+ ./fastboot flash bootloader bootloader-emmc.img
Sending 'bootloader' (1920 KB)                     OKAY [  0.127s]
Writing 'bootloader'                               OKAY [  0.104s]
Finished. Total time: 0.238s
+ ./fastboot flash uboot-env env.img
Sending 'uboot-env' (256 KB)                       OKAY [  0.023s]
Writing 'uboot-env'                                OKAY [  0.022s]
Finished. Total time: 0.052s
+ ./fastboot reboot
Rebooting                                          OKAY [  0.000s]
Finished. Total time: 0.000s
+ set +x
Power-down, remove recovery SD-CARD, and power-up the board.
Press enter to continue.

+ ./fastboot oem format
< waiting for any device >
                                                   OKAY [  0.013s]
Finished. Total time: 0.013s
+ ./fastboot flash bootloader bootloader-emmc.img
Sending 'bootloader' (1920 KB)                     OKAY [  0.127s]
Writing 'bootloader'                               OKAY [  0.104s]
Finished. Total time: 0.239s
+ ./fastboot flash uboot-env env.img
Sending 'uboot-env' (256 KB)                       OKAY [  0.023s]
Writing 'uboot-env'                                OKAY [  0.020s]
Finished. Total time: 0.052s
+ ./fastboot flash recovery_boot boot.img
Sending 'recovery_boot' (24576 KB)                 OKAY [  1.552s]
Writing 'recovery_boot'                            OKAY [  1.323s]
Finished. Total time: 2.884s
+ ./fastboot erase misc
Erasing 'misc'                                     OKAY [  0.014s]
Finished. Total time: 0.028s
+ ./fastboot reboot-fastboot
Rebooting into fastboot                            OKAY [  0.005s]
< waiting for any device >
Finished. Total time: 13.604s
+ ./fastboot flash boot_a boot.img
Sending 'boot_a' (24576 KB)                        OKAY [  1.837s]
Writing 'boot_a'                                   OKAY [  0.598s]
Finished. Total time: 2.489s
+ ./fastboot flash dtbo_a boot_dtbo.img
Sending 'dtbo_a' (41 KB)                           OKAY [  0.007s]
Writing 'dtbo_a'                                   OKAY [  0.010s]
Finished. Total time: 0.024s
+ ./fastboot flash super super.img
Invalid sparse file format at header magic
Sending sparse 'super' 1/3 (523864 KB)             OKAY [ 35.883s]
Writing 'super'                                    OKAY [  9.302s]
Sending sparse 'super' 2/3 (497864 KB)             ERROR: usb_write failed with status e00002ed
FAILED (Error reading sparse file)
fastboot: error: Command failed
pieter@RSPDW-MBP pinephone-r3 % ./flash-emmc.sh
+ ./fastboot flash gpt deploy-gpt.img
Sending 'gpt' (128 KB)                             OKAY [  0.019s]
Writing 'gpt'                                      FAILED (remote: 'No such file or directory')
fastboot: error: Command failed

Did you get this too ?

rsglobal, 2020-07-09:
@bigc0127 @dewitpj 
Thank you for testing.

It should start into bootloader fastboot mode firstly (Blank screen, just OTG flash device should be recognized by host).
There are 2 issues that prevents this from happening:
1. In the pinephone u-boot configuration OTG is turned to HOST mode only. (This forces u-boot to fallback into fastbootd)
2. Internal u-boot issue observed on OPI-WIN (Same A64 SOC) , that does not allow to enumerate device by HOST:
```
[128090.900528] usb 1-1: Device not responding to setup address.
[128091.108597] usb 1-1: Device not responding to setup address.
[128091.316246] usb 1-1: device not accepting address 39, error -71
[128091.316373] usb usb1-port1: unable to enumerate USB device
[128126.256487] usb 1-1: new high-speed USB device number 40 using xhci_hcd
[128126.388514] usb 1-1: device descriptor read/64, error -71
[128126.624309] usb 1-1: device descriptor read/64, error -71
[128126.860281] usb 1-1: new high-speed USB device number 41 using xhci_hcd
[128126.988605] usb 1-1: device descriptor read/64, error -71
[128127.224289] usb 1-1: device descriptor read/64, error -71

```

dewitpj, 2020-07-09:
Hey @rsglobal - I think I got that but it died trying to write super ?

rsglobal, 2020-07-09:
Does this happen every time?

dewitpj, 2020-07-09:
Yip - done it twice now - would it be different with the r2 image ?

rsglobal, 2020-07-09:
No, there is no sens to try with r2,
Use v0.3.0 from releases. 
r2, r3 was just temporary revisions to test UI issies.

nas00, 2020-07-09:
As I mentioned on the pine64 forum, the deploy method worked fine for both sd and emmc on my braveheart unit.

rsglobal, 2020-07-14:
I've found that hsic PLL has not initialized by u-boot on A64. That means USB should not work on any device. So, it's very weird that it works for some.

@nas00 , 
I was wondering, if you have a developer (uart) cable?
If so, can you launch minicom, and stop u-boot by pressing any key, then enter console command:
```
md 0x01c20044 1
```

nas00, 2020-07-14:
I'm afraid not but I need a new back cover anyway so I'll order serial cable at same time.

nas00, 2020-07-30:
> @nas00 ,
> I was wondering, if you have a developer (uart) cable?
> If so, can you launch minicom, and stop u-boot by pressing any key, then enter console command:
> 
> ```
> md 0x01c20000 80
> ```

The cable turned up. Disabled hardware flow control as per this [post](https://forum.pine64.org/showthread.php?tid=5026) to get it working. Here are the results:

```
U-Boot 2020.07-rc4-gdc857ca23c8b-dirty (Jul 23 2020 - 19:34:16 +0300) Allwinnery
                                                                                
CPU:   Allwinner A64 (SUN50I)                                                   
Model: SoPine with baseboard                                                    
DRAM:  2 GiB                                                                    
MMC:   mmc@1c0f000: 0, mmc@1c11000: 1                                           
Loading Environment from FAT... Unable to use mmc 1:1... In:    serial          
Out:   serial                                                                   
Err:   serial                                                                   
Allwinner Sun4I USB PHY driver loaded                                           
sun4i_usb_phy_xlate: phy_id = 0                                                 
Allwinner mUSB OTG (Peripheral)                                                 
Net:   phy interface7                                                           
eth0: ethernet@1c30000                                                          
Warning: usb_ether using MAC address from ROM                                   
, eth1: usb_ether                                                               
Hit any key to stop autoboot:  0                                                
=> md 0x01c20000 80                                                             
01c20000: 90001031 00000000 00035514 00000000    1........U......               
01c20010: 03006207 00000000 03006207 00000000    .b.......b......               
01c20020: 00001000 00000000 90041811 00041811    ................               
01c20030: 03006207 00000000 03006207 00000000    .b.......b......               
01c20040: 00000515 03001300 03006207 90002d00    .........b...-..               
01c20050: 00020102 00003180 01000000 00000001    .....1..........               
01c20060: 00824500 00000001 00000020 00010000    .E...... .......               
01c20070: 00000000 00000000 00000000 00000000    ................               
01c20080: 00000001 00000000 80000000 00000000    ................               
01c20090: 8100000b 00000000 00000000 00000000    ................               
01c200a0: 00000000 00000000 00000000 00000000    ................               
01c200b0: 00000000 00000000 00000000 00000000    ................               
01c200c0: 00000000 00000000 00000000 00000101    ................               
01c200d0: 00000000 00000000 00000000 00000000    ................               
01c200e0: 00000000 00000000 00000000 00000000    ................               
01c200f0: 00000000 80100000 ccca0000 80000000    ................               
01c20100: 00000000 00000000 00000000 00000000    ................               
01c20110: 00000000 00000000 00000000 00000000    ................               
01c20120: 00000000 00000000 00000000 00000000    ................               
01c20130: 00000000 00000000 00000000 00000000    ................               
01c20140: 00000000 00000000 00000000 00000000    ................               
01c20150: 00000000 00000000 00000000 81000002    ................               
01c20160: 00000000 00000000 00000000 00000000    ................               
01c20170: 00000000 00000000 00000000 00000000    ................               
01c20180: 00000000 00000000 00000000 00000000    ................               
01c20190: 00000000 00000000 00000000 00000000    ................               
01c201a0: 00000000 00000000 00000000 00000000    ................               
01c201b0: 80000000 00000000 00000000 00000000    ................               
01c201c0: 00000000 00000000 00000000 00000000    ................               
01c201d0: 00000000 00000000 00000000 00000000    ................               
01c201e0: 00000000 00000000 00000000 00000000    ................               
01c201f0: 00000000 00000000 00000000 00000000    ................               
=> 
```

jholtiv, 2020-09-17:
@dewitpj The latest build works with emmc install. please close this issue if it works for you

Shatur95, 2021-02-19:
Tested, the latest build (v0.5.1) works fine.

closed 2021-05-06 by rsglobal