GloDroid/glodroid_manifest #33: Pinephone Wifi not connecting to encrypted SSID

dewitpj, 2020-07-08:
Hi,

I have a SSID with a pretty long key - 26 chars to be exact :)

I could connect to this SSID using UBPORTS but in GloDriod it goes disabled. I am not sure what more debugging to send you, please ask :)

SSID is using WPA Personal, WPA2 Only - with CCMP encryption.

Thanks !

dewitpj, 2020-07-08:
Worth noting - WiFi works when connecting to an unencrypted SSID

philnik777, 2020-07-25:
I could not reproduce the problem. I used "123456789012345678901234567890". What version do you use? I used 0.3.0. Is it some combination of characters? It would be great if you could try that or just post the password if it is not important.

philnik777, 2020-07-25:
Did you use the same SSID? If not it may be that as well.

dewitpj, 2020-07-25:
> Did you use the same SSID? If not it may be that as well.

I created a new SSID....originally called...."Test"

It might be the encryption types - I will post those shortly

dewitpj, 2020-07-25:
[image]

Oh and yes, this was on 0.3.0 :)

rsglobal, 2020-08-07:
Please test if this issue is fixed:
https://github.com/GloDroid/glodroid_manifest/issues/58#issuecomment-670354906

herbsmn, 2022-11-04:
I was able to connect to an encrypted SSID on my PinePhone running v0.8.1. I suggest closing this issue.

closed 2022-11-04 by rsglobal