GloDroid/glodroid_manifest #34: PINEPHONE: Pulsing display

nas00, 2020-07-08:
The screen constantly pulses below approx 60% brightness. Pulsing becomes more noticeable as brightness is reduced to minimum.

dewitpj, 2020-07-08:
I think I saw this on an earlier version - I now have a "band of white" on my display - images to follow

nas00, 2020-07-08:
I haven't seen whites bands - it's more like a flicker as if the refresh rate is wrong. Seen similar issue on early linux releases.

rakor, 2020-07-11:
Here too. The screen is flickering and all hard contrasts have like vertical shadows. So a white box in the middle of the screen gives light lines to the top and the bottom of the screen.

rsglobal, 2020-07-14:
This patch describes something similar:
https://lore.kernel.org/patchwork/patch/1267690/

I still waiting for device, so I can't check now.

nas00, 2020-07-14:
I've seen the bluetooth issue on debian - every input with bt device caused brightness to increase momentarily. The issue here is a faster flicker - I suppose could be related to the wifi so maybe flipping the relevant kill switches would confirm?

rsglobal, 2020-07-15:
I've reduced wifi voltage to 1.8:
Can anybody check it
https://drive.google.com/file/d/1MqHAg_g9Jm_5vI2RLtByuKy3_tncvkY7/view?usp=sharing
?

dewitpj, 2020-07-15:
Downloading - will advise

dewitpj, 2020-07-15:
I still have the white bars, no flicker as yet

rsglobal, 2020-07-15:
@dewitpj ,
Can you attach a picture here of the bars?

dewitpj, 2020-07-15:
Sorry, I did forget to do that:

It’s only when a white screen is present on a dark background 

Thanks!

dewitpj, 2020-07-15:
Seems email attachments don't work :(

[image]

nas00, 2020-07-20:
> I've reduced wifi voltage to 1.8:
> Can anybody check it
> 

It didn't work but it reminded of this display voltage issue:

https://mobile.twitter.com/RealDanct12/status/1232456700868710400

anibyl, 2020-07-20:
I'm experiencing the same issue.

The temporary "fix" I'm using: brightness ~50% + dark theme.

nas00, 2020-07-20:
Are those white bars present in other OSes? They look a bit like LCD hardware issues reported by a few owners.

> Seems email attachments don't work :(
>

dewitpj, 2020-07-20:
> Are those white bars present in other OSes? They look a bit like LCD hardware issues reported by a few owners.
> 
> > Seems email attachments don't work :(

Nope, no lines on other OSs :(

anibyl, 2020-07-20:
> Nope, no lines on other OSs :(

Can confirm, these are the lines from bright elements on the screen, like icons and buttons.

Exonip, 2020-08-03:
> > Nope, no lines on other OSs :(
> 
> Can confirm, these are the lines from bright elements on the screen, like icons and buttons.

Can confirm aswell. Any updated yet?

rsglobal, 2020-08-03:
This and many other hardware issues solved by @megous in his kernel v5.8 integration branch. (https://github.com/megous/linux/tree/orange-pi-5.8)
Currently we are integrating in into the GloDroid.

rsglobal, 2020-08-07:
Please test if this issue is fixed:
https://github.com/GloDroid/glodroid_manifest/issues/58#issuecomment-670354906

closed 2020-08-07 by rsglobal