GloDroid/glodroid_manifest #35: PINEPHONE : Google Play Store not logging in

dewitpj, 2020-07-08:
Hi,

Yes, this is Google Play Store and yes, not what Pinephone was made for...but...

After installing all of the play store "things", I get the Play Store app on my phone. Once I run it, it prompts for a login and it doesn't matter what I do, it never progresses from that stage.

I do however note on the serial console that avc is blocking a few events/file accesses for Google Play Store - would this contrib. to the issue ? Is there an easy way to disable SELinux ? I have tried the adb setenforce 0 "thing" and it doesn't seem to make a difference - even to the messages.

rsglobal, 2020-07-09:
Hi. 

Google play store and other apps from GMS (https://www.android.com/gms/) are not FOSS.
To support them legally you have to get approval from Google (Not for free).
Maybe someday official PinePhone team will go this way, but currently there is no such plans AFAIK.

closed 2020-07-09 by rsglobal

nas00, 2020-07-09:
You could try the micro g instead but think you'd need to get signature spoofing working somehow.

dewitpj, 2020-07-09:
Thanks - will look into that :)