GloDroid/glodroid_manifest #39: PINEPHONE: SD card recovery won't boot

XenonFiber, 2020-07-13:
I formatted my card with deploy-sd.img at first, and tried again with deploy-sd-for-emmc.img, and both times my phone couldn't boot from it. Holding power does nothing and the LED stays unlit. The phone can boot fine from both eMMC and SD, and I've tested this with UBports on eMMC, and Jumpdrive on SD. I used the same tool to format it for Jumpdrive as I did for the recovery image (gnome-disk-utility 3.28.3).

ADB doesn't detect a device with any combination of power button presses, so it doesn't appear to be booting at all.

rsglobal, 2020-07-13:
Deploy images should run fastboot usb peripheral, after that flash-sd.sh script should work.
Can you provide your host PC `dmesg` logs?

XenonFiber, 2020-07-13:
Sure, where would they be stored?

XenonFiber, 2020-07-13:
Oh, sorry. I've got them here:

`[156149.378414] usb 3-3: new high-speed USB device number 8 using xhci_hcd
[156149.506472] usb 3-3: device descriptor read/64, error -71
[156149.746423] usb 3-3: device descriptor read/64, error -71
[156149.982426] usb 3-3: new high-speed USB device number 9 using xhci_hcd
[156150.110438] usb 3-3: device descriptor read/64, error -71
[156150.350512] usb 3-3: device descriptor read/64, error -71
[156150.458490] usb usb3-port3: attempt power cycle
[156151.110425] usb 3-3: new high-speed USB device number 10 using xhci_hcd
[156151.110451] usb 3-3: Device not responding to setup address.
[156151.110502] debugfs: Directory '03' with parent 'devices' already present!
[156151.318465] usb 3-3: Device not responding to setup address.
[156151.318508] debugfs: Directory '03' with parent 'devices' already present!
[156151.526435] usb 3-3: device not accepting address 10, error -71
[156151.526491] debugfs: Directory '03' with parent 'devices' already present!
[156151.658445] usb 3-3: new high-speed USB device number 11 using xhci_hcd
[156151.658476] usb 3-3: Device not responding to setup address.
[156151.658525] debugfs: Directory '03' with parent 'devices' already present!
[156151.866469] usb 3-3: Device not responding to setup address.
[156151.866516] debugfs: Directory '03' with parent 'devices' already present!
[156152.074437] usb 3-3: device not accepting address 11, error -71
[156152.074476] usb usb3-port3: unable to enumerate USB device
`

XenonFiber, 2020-07-14:
I switched USB ports and got a separate error:

[179185.213597] usb 1-1: new high-speed USB device number 10 using xhci_hcd
[179190.014094] usb 1-1: Device not responding to setup address.
[179195.022731] usb 1-1: Device not responding to setup address.
[179195.229782] usb 1-1: device not accepting address 10, error -71
[179195.633792] usb 1-1: new high-speed USB device number 11 using xhci_hcd
[179200.434406] usb 1-1: Device not responding to setup address.
[179205.446047] usb 1-1: Device not responding to setup address.
[179205.657995] usb 1-1: device not accepting address 11, error -71
[179205.658047] usb usb1-port1: unable to enumerate USB device

rsglobal, 2020-07-14:
It similar to errors I've got on my opi-win.
It looks like some of A64 SOCs have working u-boot usb gadget and some of them do not.

XenonFiber, 2020-07-14:
Oh, huh. Do you know of a method to bypass the need for this? Maybe some way to directly install off an SD card or eMMC without needing a host PC?

rsglobal, 2020-07-14:
You can try luck using gensdimg.sh tool: https://github.com/GloDroid/glodroid_device/blob/master/platform/tools/gensdimg.sh
Do not forget to run `sync` shell command before removing the media/reboot.

XenonFiber, 2020-07-14:
I was able to get to an Android bootloop, so that's an improvement to it not powering on.

To document the exact procedure for anyone reading this later, I created the script in the GloDroid/out/target/product/pinephone/images folder, marked it as executable, and ran it. Once the script completed, I restored the newly generated sdcard.img file to my SD card and ran it on my Pinephone.

Is there a next part of the procedure I missed, or is this a sign that it's not going to be able to boot correctly as of right now? It looks like it's recognized by ADB, but it's unable to be listed correctly (or used) with the error, "no permissions (user in plugdev group; are your udev rules wrong?)."

rsglobal, 2020-07-14:
Is v0.2.0 .sd image working for you?

XenonFiber, 2020-07-14:
Yup, just tried it and it booted just fine. So v0.2.0 works, but v0.3.0 bootloops.

If you need any more info about my specific Pinephone or want the results from a test, I can definitely send then your way, but you'll probably the more important details from the v1.2 revision's wiki page: https://wiki.pine64.org/index.php?title=PinePhone_v1.2

Thanks a lot for helping me get something running here, I appreciate it.

XenonFiber, 2020-07-14:
I'm noticing a lot of full OS crashing in v0.2.0. Was this an issue in that version, or is it an issue with my specific setup?

rsglobal, 2020-07-16:
@XenonFiber ,

v0.3.0 has 256mb reserved for graphics, when 0.2.0 has just 128, so it should be less glitches.
Can you check if adb works on your board with 0.2.0?

rsglobal, 2020-07-16:
@XenonFiber ,

Can you try to run 
`sudo ./gensdimg.sh /dev/sdXXX`, where /dev/sdXXX is your target sdcard?

XenonFiber, 2020-07-16:
Yeah, ADB works fine on v0.2.0. I also fixed an issue with the host PC involving the USB connection that was preventing me from using ADB properly, so I retested the original instructions to make sure that wasn't the cause and it was almost definitely unrelated. Still can't boot with deploy-sd.img.

I ran the command, reinstalled the SD card, and got into an Android bootloop again. So far, flashing the card with v0.2.0 has been the only successful boot. Installing the OS to the card directly has always gotten a bootloop, and neither of the deploy images will power up at all.

muellermartin, 2020-07-23:
I also have this issue with a PinePhone (Braveheart edition) and gd-pinephone-v0.3.0 release. I tried to flash both `deploy-sd.img` and `deploy-sd-for-emmc.img` to a microSD card by using `dd` on macOS, but in both cases the phone did not boot (screen stays black and no LED lights up). I will try the deployment strategy with Jumpdrive like mentioned in issue #32

anjandev, 2020-07-30:
I also have this issue with 0.3.0 release. I tried `sudo ./gensdimg.sh /dev/sdXXX`, I got a boot loop. I flashed v0.2.0 to an sd card and I was able to boot.

I have a braveheart edition pinephone.

rsglobal, 2020-07-30:
Unfortunately USB-OTG can not work in u-boot for some devices. And this is low-level problem in some SOCs or ROM code and will be investigated, but I do not expect results soon.
I'll check why does gensdimg.sh script produces incorrect images nearest time.

nas00, 2020-08-01:
I tried gensdimg.sh and it created an 8gb file which I flashed to an sd card with etcher. Boots up fine too.

[gensdimg.log](https://github.com/GloDroid/glodroid_manifest/files/5010461/gensdimg.log)

jholtiv, 2020-09-17:
@XenonFiber please try the latest build and it should work, Mark this as closed if it works. Otherwise update us

uncivilgentleman, 2020-10-22:
I don't exactly know what I'm doing but I'm also unable to get the deploy-sd-for-emmc.img to do anything but turn the Pinephone's red light on.  I tried using gensdimg.sh to create an 8.1GB image then wrote that image to a microsd card.  That got the Pinephone to boot to an Android logo but nothing else.

muellermartin, 2021-01-14:
For me it now worked with the latest pre-release GloDroid v0.5.1 (24-09-2020-17h09m_release_v0.5.1_pinephone-userdebug.tar.gz) on macOS.

Steps:
1. Change into directory with images
2. Write image to SD card: `sudo dd if=deploy-sd-for-emmc.img of=/dev/rdisk2 bs=1m` (note that device `rdisk2` might be different for you – find out with `diskutil list`)
2. Install `android-platform-tools` with Homebrew: `brew install --cask android-platform-tools`
3. Replace ELF binaries with symlink to binaries from Android platform tools package: `mv fastboot fastboot.bak && ln -s $(which fastboot) fastboot` and `mv mke2fs mke2fs.bak && ln -s $(which mke2fs) mke2fs`
4. Run `./flash-emmc.sh` and follow instructions

vinyanalista, 2021-08-13:
I tried GloDroid v0.6.1 today.

Installing to the SD card didn't work, neither using provided Android Tools binaries nor system's Android Tools binaries. My PinePhone keeps rebooting forever with the status LED in red.

Installing to the eMMC worked using provided Android Tools binaries.

I'm using openSUSE Leap 15.3 (64-bit Linux) and my PinePhone is the postmarketOS Community Edition with 3GB RAM and 32Gb eMMC.

coeusite, 2021-11-25:
> I tried GloDroid v0.6.1 today.
>
> Installing to the SD card didn't work, neither using provided Android Tools binaries nor system's Android Tools binaries. My PinePhone keeps rebooting forever with [...]

Same issue here. I need to manually boot to fastboot by pressing down+power in the middle of the flashing process. 

`fastboot reboot-fastboot` does not work on my end.

rsglobal, 2021-11-25:
Please try to follow this guide:
https://www.youtube.com/watch?v=ZdmBq0rFPa8&ab_channel=LINMOBnet

rsglobal, 2023-03-21:
Closing due to inactivity. If this is still relevant, please reopen it [HERE](https://github.com/GloDroidCommunity/pine64-pinephone/issues):

closed 2023-03-21 by rsglobal