GloDroid/glodroid_manifest #40: WiFi Aware Support

mbalex99, 2020-07-16:
Hi,

I'm quite interested in this project but I was wondering:

1. Is there WiFi Aware https://www.wi-fi.org/discover-wi-fi/wi-fi-aware support? 
2. If no, how could we contribute to add WiFi Aware Support especially for the Pi -type of phones?

Max

rsglobal, 2020-07-16:
Hello Max,

Currently there is no support for wifi-aware.

According to:
https://source.android.com/devices/tech/connect/wifi-aware

It would require WiFi HAL, that is missing for PP an other sunxi devices out of the box.

Regards,
Roman

rsglobal, 2020-07-16:
There is some implementation for WIFI HAL for Broadcom
https://android.googlesource.com/platform/hardware/broadcom/wlan/+/refs/heads/master/bcmdhd/wifi_hal/

So you can try luck to bring it on RPI4 first.

rsglobal, 2022-11-20:
closing as stale

closed 2022-11-20 by rsglobal