GloDroid/glodroid_manifest #42: Instructions missing steps, rPi4 doesn't build u-boot (Ubuntu 14.04 or U 20.04)

myne123, 2020-07-19:
The listed steps don't work with a fresh install of Ubuntu.
14.04 is recommended by Google in the linked build page. 20.04 has the same issues.

Missing: 
apt-get install git gettext python3-pip libncurses5
 ( I think I forgot one or two too. Please try)
Not needed: 
apt-get install repo

u-boot:
make: Entering directory `/home/m/GloDroid/GloDroid/external/u-boot'
Makefile:37: *** missing separator. Stop.

Line 37: undefine MK_ARCH

That's it. I'm quite amateur, so it's not something I can fix.

rsglobal, 2020-07-21:
@myne123 ,
Thank you for the report.

@MaksymPrymierov ,
Could you install clean Ubuntu 20.04 on virtualbox on BuildServer and check required dependencies?

MaksymPrymierov, 2020-07-23:
@rsglobal Yes, I'll check it out.

MaksymPrymierov, 2020-08-14:
@myne123 
I have tried to get and build Android on fresh Ubuntu 20.04. I have found missing packages that were attached by you, but I did not face your error.
Could you please attach the full build log file?

@rsglobal 
I also created a pull request with updated steps for installing packages for Android build. https://github.com/GloDroid/glodroid_manifest/pull/65

rsglobal, 2020-09-03:
Merged into next.

closed 2020-09-03 by rsglobal