GloDroid/glodroid_manifest #44: PINEPHONE: Change default wallpaper

rsglobal, 2020-07-21:
AOSP supply default wallpaper as large resolution image.
Since PinePhone and PineTab display resolution is known, we can scale-down default wall-paper to match that resolution.

This will reduce RAM usage and increase UI performance, since SurfaceFlinger won't scale wallpaper every frame anymore.

AleksandrBulyshchenko, 2020-07-22:
As far as I ca see
PinePhone has 720x1440 display;
PineTab - 800x1280.

rsglobal, 2020-07-23:
I've tried to change wallpaper to match screen resolution (1024x600), but I still see larger frame in `dumpsys SurfaceFlinger'.
```
-------------------------------------------------------------------------------------------
 Layer name
           Z |  Window Type |  Comp Type |  Transform |   Disp Frame (LTRB) |          Source Crop (LTRB)
-------------------------------------------------------------------------------------------
 com.android.systemui.ImageWallpaper#0
  rel      0 |         2013 |     CLIENT |          0 |    0    0 1024  600 |    0.0    0.0 1747.0 1024.0
- - - - - - - - - - - - - - - -- - - - - - - - - - - - - - - -- - - - - - - - - - - - - - -
 com.android.launcher3/com.android.launcher3.Launcher#0
  rel      0 |            1 |     CLIENT |          0 |    0    0 1024  600 |    0.0    0.0 1024.0  600.0
- - - - - - - - - - - - - - - -- - - - - - - - - - - - - - - -- - - - - - - - - - - - - - -
 StatusBar#0
  rel      0 |         2000 |     CLIENT |          0 |    0    0 1024   24 |    0.0    0.0 1024.0   24.0
- - - - - - - - - - - - - - - -- - - - - - - - - - - - - - - -- - - - - - - - - - - - - - -
 NavigationBar0#0
  rel      0 |         2019 |     CLIENT |          0 |    0  552 1024  600 |    0.0    0.0 1024.0   48.0
- - - - - - - - - - - - - - - -- - - - - - - - - - - - - - - -- - - - - - - - - - - - - - -
```
UI is much smoother now, but still it makes sense to fully eliminate the scaling, @AleksandrBulyshchenko , can someone investigate why it happens?

MaksymPrymierov, 2020-07-23:
@rsglobal I'm going to investigate this.

rsglobal, 2020-08-15:
This feature is included into the integration branch and will be available in the next release.
@MaksymPrymierov , Thank you for your effort.

closed 2020-08-15 by rsglobal