GloDroid/glodroid_manifest #47: Allow overriding modifiers for graphic buffers

rsglobal, 2020-07-23:
Currently GloDroid graphics pipeline supports only LINEAR buffers.
Since tiled buffers can increase UI performance (increase GPU cache hits), it makes sense to try to enable it.

After fixing mesa issue with modifiers: https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/6045 ,
now it possible to use different (mixed) buffer modifiers.

Experimental results showed that using TILED modifiers for all buffers except HWFB (Composition on client side) gives huge performance gain comparing to using linear buffers (Composition on client side case).
UI performance increase feels like about 4x on v3d GPU, and about 1.5x-2x on MALI-400.

Action items:
1. Import modifier by drm_hwomposer generic importer.
2. Allow to override modifiers with gbm_gralloc, (enable [AFBC](https://www.kernel.org/doc/html/latest/gpu/afbc.html), different tiling models, etc.). Similar is hard-coded here: https://gitlab.collabora.com/spurv/gbm_gralloc/-/commit/c61679f897f76ddbab2ff3dd11a7cde9847fbfd9
Overriding should be enabled via properties for different buffer usage flags separately.

rsglobal, 2020-07-23:
Pinephone build using tiled graphic buffers only:
https://drive.google.com/file/d/1Oj9tT9Pz1z67R5wsxzwt8pm0u0pgl2Qp/view?usp=sharing

@nas00, could you check how UI performance changed on PP?

nas00, 2020-07-23:
Yes, it feel likes 2-3x speed increase in places - heres a [quick videi](https://www.dropbox.com/s/51t5vjnr2y4hm76/glodroid-test.mp4?dl=1)

nas00, 2020-07-23:
However, video playback appears to be broken - youtube shows black window with sound in browser.

nas00, 2020-08-08:
Youtube videos shot in horizontal aspect ratio work at 240p or below. Portrait videos work at 360p or below. Going above those limits results in frozen/flickering picture or black frame.

rsglobal, 2020-08-08:
@nas00 , thank you for testing. Can you check today's integration build. Youtube issue should be fixed.

nas00, 2020-08-08:
360p now works reliably. 480p and 720p mostly give  'An error occurred' message or frozen/black frame.  1080p works half of time.

rsglobal, 2020-08-10:
Thanks, I've applied some fixes and updated integration build.

nas00, 2020-08-10:
Video playback seems to work up to 720p now. Tried 1080p but frozen frame - possibly related to pinephone overheating in this hot weather so will test when cooler. Still no audio. Also, notice every video gives repeating messages like these now:

08-10 23:01:40.205   304 11031 D GRALLOC-GBM: handle 0xf739b06f2f30, hnd 0xf739b06f2f30, usage 0x33
08-10 23:01:40.205   304 11031 I GRALLOC-GBM: lock bo 0xf739b06f49d0, cnt=0, usage=33

rsglobal, 2020-08-11:
Audio functionality restored.
logcat messages is debug messages, I will silence them later.

nas00, 2020-08-11:
Good job - 1080p and audio both working fine now.

rsglobal, 2020-08-18:
After more investigation, it turns out that using modifiers does not offer much performance increasing on lima, we will improve GPU off-loading instead.
As for AFBC, separated ticket will be created when we will be ready for it.

closed 2020-08-18 by rsglobal