GloDroid/glodroid_manifest #48: Build fails with error in gatekeeper

philnik777, 2020-07-23:
I tried to build the project. Yesterday everything worked fine. Today I synced the repo and now it gives me this error:

`============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=10
TARGET_PRODUCT=aosp_arm
TARGET_BUILD_VARIANT=eng
TARGET_BUILD_TYPE=release
TARGET_ARCH=arm
TARGET_ARCH_VARIANT=armv7-a-neon
TARGET_CPU_VARIANT=generic
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-5.4.0-42-generic-x86_64-Ubuntu-18.04.4-LTS
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=QQ3A.200605.002.A1
OUT_DIR=out
============================================
[ 61% 565/913] including device/glodroid/common/Android.mk ...
FAILED: 
build/make/core/base_rules.mk:325: error: device/generic/goldfish/gatekeeper: MODULE.TARGET.SHARED_LIBRARIES.gatekeeper.ranchu already defined by device/generic/goldfish/gatekeeper.
16:39:52 ckati failed with: exit status 1
build/make/core/main.mk:21: recipe for target 'run_soong_ui' failed
make: *** [run_soong_ui] Error 1`

rsglobal, 2020-07-23:
It seems you forgot to select your device with `lunch` command.

philnik777, 2020-07-23:
Thank you very much. I would love to contribute to the GloDroid but don't really understand the android project. I already searched on the web but did not really find anything. Do you have any tips for me where to find information? (other than looking at the source for hours). Programming knowledge is not the problem I think. I know some C++, Java and Python and how hardware works (to some extent).

closed 2020-07-23 by philnik777

rsglobal, 2020-07-23:
1. Most recent and relevant information can be found in git commits/and on AOSP gerrit. 
https://android-review.googlesource.com/dashboard/self

2. https://source.android.com/ is a good resource to start. There are a lot of useful information there.
 
If you want to contribute some particular feature, just describe it and we will suggest where to start learning.