GloDroid/glodroid_manifest #49: Questions on AOSP + Pinephone

thunderbiscuit, 2020-07-25:
Hey guys,

I know this is probably not the perfect place to ask this but I'm just getting into this and don't really know the right forums/chats to go to. Feel free to redirect me somewhere else!

I am trying to wrap my head around the Android Open Source Project and the pieces that come together to make it work on specific devices. The image attached is my current mental model of how Android can be built. I am trying to

1. refine/restructure the mental model it if it needs it, and

2. understand better if "vendor blobs" are the only thing one needs to make AOSP work on any given device or if more is needed.

Eventually the bigger picture question I have is whether one could build "vanilla" AOSP from source and use it on a Pinephone or whether other pieces are still missing in that architecture to make it work that I have not understood/accounted for. Implicit in this is the question of whether those "vendor blobs" are all available for the Pinephone.

Any help or readings or docs you can send me to to sharpen my understanding of the pieces of the puzzle this would be great!

[image]

codiflow, 2020-07-26:
Maybe this can help you getting more information on the binary blobs:
https://www.pine64.org/2020/01/24/setting-the-record-straight-pinephone-misconceptions/

thunderbiscuit, 2020-07-26:
I see. Cheers for the link! Sometimes those little blog posts are too small to be picked up by the search engines, or at least don't come up on the first page, so they're hard to find unless someone points them out to ya. So indeed the answer is yes, all necessary binary blobs are available for the Pinephone!

Am I correct in understanding that this really is all you'd need to build vanilla AOSP and target the Pinephone hardware? The fact that Glodroid is a work in progress and I was reading that the camera and telephony are not working yet tells me there is much more to it than my little diagram implies.

What are some of the roadblocks in between us and a plain AOSP working on the Pinephone? Sorry if this is too nooby of a question haha. I'll be trying to build the vanilla AOSP for a common target (maybe like a Pixel) over the next few weeks, and so I might have better/more structured questions then. Feel free to close this issue if you don't want it to pollute the main repo, and thanks for the help! 

(maybe one more question: where do you guys who work on AOSP hang out in terms of chat rooms? Anywhere I should go for more of these questions instead of opening issues? I posted on the Pine64 forums but didn't get any traction there...)

kuleszdl, 2020-07-28:
I've been building and running AOSP on Nexus und Pixel devices used to work "somehow" in the past but has been getting harder and harder due to the way to vendor blobs had to be modified after extracting them. If you're interested in some details you can check out this article I have written a while ago:

https://www.howtoforge.com/tutorial/building-and-flashing-a-secured-aosp-build-with-verified-boot-and-separate-lockscreen-password-for-the-nexus-5x/

Nowadays, you cab get the best and easiest close-to-AOSP experience by running GrapheneOS on one of the supported pixel phones.

Regarding the Pinephone and AOSP I'm not an expert. My understanding of the issue is that you need additional drivers and kernel patches for Android - not just a mainline kernel. And I'm not talking about the vendor blobs here. You can find these patches for the pinephone in the `kernel_sunxi` branch of this project I believe:

https://github.com/GloDroid/glodroid_forks/tree/kernel-sunxi-next