GloDroid/glodroid_manifest #50: u-boot won't build. Rpi4

myne123, 2020-07-29:
I reported it in the other thread, but it seems you only saw the dependency issue.
It does this with both the main and next branch. 
Everything else seems to build.

[  0% 451/90066] Building U-Boot:
FAILED: out/target/product/rpi4/obj/UBOOT_OBJ/u-boot.bin
/bin/bash -c "(echo \"TARGET_PRODUCT = \" rpi4: ) && (mkdir -p out/target/product/rpi4/obj/UBOOT_OBJ ) && (PATH=/usr/bin:/bin:\$PATH ARCH=arm64 CROSS_COMPILE=\$(readlink -f prebuilts/gcc/linux-x86/aarch64/gcc-linaro-aarch64-linux-gnu/bin/aarch64-linux-gnu-) BL31=\$(readlink -f out/target/product/rpi4/obj/ATF_OBJ/rpi4/debug/bl31.bin) make -j1 -C external/u-boot O=\$(readlink -f out/target/product/rpi4/obj/UBOOT_OBJ) rpi_4_defconfig ) && (PATH=/usr/bin:/bin external/u-boot/scripts/kconfig/merge_config.sh -m -O out/target/product/rpi4/obj/UBOOT_OBJ/ out/target/product/rpi4/obj/UBOOT_OBJ/.config device/glodroid/platform/common/uboot.config device/glodroid/platform/common/broadcom/uboot.config device/glodroid/platform/common/uboot-sd.config ) && (PATH=/usr/bin:/bin:\$PATH ARCH=arm64 CROSS_COMPILE=\$(readlink -f prebuilts/gcc/linux-x86/aarch64/gcc-linaro-aarch64-linux-gnu/bin/aarch64-linux-gnu-) BL31=\$(readlink -f out/target/product/rpi4/obj/ATF_OBJ/rpi4/debug/bl31.bin) make -j1 -C external/u-boot O=\$(readlink -f out/target/product/rpi4/obj/UBOOT_OBJ) olddefconfig ) && (PATH=/usr/bin:/bin:\$PATH ARCH=arm64 CROSS_COMPILE=\$(readlink -f prebuilts/gcc/linux-x86/aarch64/gcc-linaro-aarch64-linux-gnu/bin/aarch64-linux-gnu-) BL31=\$(readlink -f out/target/product/rpi4/obj/ATF_OBJ/rpi4/debug/bl31.bin) make -j1 -C external/u-boot O=\$(readlink -f out/target/product/rpi4/obj/UBOOT_OBJ) KCFLAGS=\"-fgnu89-inline \" ) && (cp out/target/product/rpi4/obj/UBOOT_OBJ/u-boot.bin out/target/product/rpi4/obj/UBOOT_OBJ/u-boot.bin.sd )"
TARGET_PRODUCT =  rpi4:
make: Entering directory `/home/m/GloDroid/external/u-boot'
Makefile:37: *** missing separator.  Stop.
make: Leaving directory `/home/m/GloDroid/external/u-boot'
22:46:53 ninja failed with: exit status 1

#### failed to build some targets (47 seconds) ####

rsglobal, 2020-07-29:
I saw your issue, but I have no idea why is it happen. I build it for rpi4 many times also from scratch without any issues.
Can you share your PC model.
Also getting full build log will be helpful to understand what's going wrong. 
`make images -j1 2>&1 |  tee build.log`

myne123, 2020-07-30:
It's a VM using Hyper-V on a Z390/Kaby Lake if that's relevant.

Result of log: 

============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=10
TARGET_PRODUCT=rpi4
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a53
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=cortex-a15
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-4.4.0-148-generic-x86_64-Ubuntu-14.04.6-LTS
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=QQ3A.200705.002
OUT_DIR=out
============================================
ninja: no work to do.
[  0% 1/450] glob AppWidgetHostTest.prof
[  0% 2/450] glob AppWidgetProvider.prof
[  0% 3/450] glob AppcompatPhotoViewerSample.prof
[  0% 4/450] glob BackupRestoreConfirmation.prof
[  1% 5/450] glob BasicDreams.prof
[  1% 6/450] glob BasicSmsReceiver.prof
[  1% 7/450] glob BlockedNumberProvider.prof
[  1% 8/450] glob BluetoothMidiService.prof
[  2% 9/450] glob BookmarkProvider.prof
[  2% 10/450] glob Browser2.prof
[  2% 11/450] glob BuiltInPrintService.prof
[  2% 12/450] glob CalendarProvider.prof
[  2% 13/450] glob CallLogBackup.prof
[  3% 14/450] glob CaptivePortalLogin.prof
[  3% 15/450] glob CarSystemUI.prof
[  3% 16/450] glob CarrierConfig.prof
[  3% 17/450] glob CarrierDefaultApp.prof
[  4% 18/450] glob CellBroadcastReceiver.prof
[  4% 19/450] glob CertInstaller.prof
[  4% 20/450] glob CompanionDeviceManager.prof
[  4% 21/450] glob ContactsProvider.prof
[  4% 22/450] glob DeskClock.prof
[  5% 23/450] glob DevCamera.prof
[  5% 24/450] glob DocumentsUI.prof
[  5% 25/450] glob DownloadProviderUi.prof
[  5% 26/450] glob DownloadProvider.prof
[  6% 27/450] glob DozeTest.prof
[  6% 28/450] glob EasterEgg.prof
[  6% 29/450] glob EmergencyInfo.prof
[  6% 30/450] glob ExamplePlugin.prof
[  6% 31/450] glob ExtServices.prof
[  7% 32/450] glob ExternalStorageProvider.prof
[  7% 33/450] glob FakeOemFeatures.prof
[  7% 34/450] glob FixVibrateSetting.prof
[  7% 35/450] glob FrameSequenceSample.prof
[  8% 36/450] glob FusedLocation.prof
[  8% 37/450] glob GL2CameraEye.prof
[  8% 38/450] glob GL2JNI.prof
[  8% 39/450] glob GL2Java.prof
[  8% 40/450] glob GLDual.prof
[  9% 41/450] glob GLJNI.prof
[  9% 42/450] glob GLPerf.prof
[... 398 lines cut ...]
[ 98% 441/450] glob test-vendor-java.prof
[ 98% 442/450] glob updatable-media.prof
[ 98% 443/450] glob vendor.nxp.nxpese-V1.0-java-shallow.prof
[ 98% 444/450] glob vendor.nxp.nxpese-V1.0-java.prof
[ 98% 445/450] glob vendor.nxp.nxpnfc-V1.0-java-shallow.prof
[ 99% 446/450] glob vendor.nxp.nxpnfc-V1.0-java.prof
[ 99% 447/450] glob vmtests-mains.prof
[ 99% 448/450] glob voip-common.prof
[ 99% 449/450] glob wifi/java/**/*.java
No need to regenerate ninja file
No need to regenerate ninja file
No need to regenerate ninja file
Starting ninja...
[  0% 450/90062] build out/target/product/rpi4/symbols/apex/com.android.runtime
[  0% 451/90062] Building U-Boot:
FAILED: out/target/product/rpi4/obj/UBOOT_OBJ/u-boot.bin
/bin/bash -c "(echo \"TARGET_PRODUCT = \" rpi4: ) && (mkdir -p out/target/product/rpi4/obj/UBOOT_OBJ ) && (PATH=/usr/bin:/bin:\$PATH ARCH=arm64 CROSS_COMPILE=\$(readlink -f prebuilts/gcc/linux-x86/aarch64/gcc-linaro-aarch64-linux-gnu/bin/aarch64-linux-gnu-) BL31=\$(readlink -f out/target/product/rpi4/obj/ATF_OBJ/rpi4/debug/bl31.bin) make -j1 -C external/u-boot O=\$(readlink -f out/target/product/rpi4/obj/UBOOT_OBJ) rpi_4_defconfig ) && (PATH=/usr/bin:/bin external/u-boot/scripts/kconfig/merge_config.sh -m -O out/target/product/rpi4/obj/UBOOT_OBJ/ out/target/product/rpi4/obj/UBOOT_OBJ/.config device/glodroid/platform/common/uboot.config device/glodroid/platform/common/broadcom/uboot.config device/glodroid/platform/common/uboot-sd.config ) && (PATH=/usr/bin:/bin:\$PATH ARCH=arm64 CROSS_COMPILE=\$(readlink -f prebuilts/gcc/linux-x86/aarch64/gcc-linaro-aarch64-linux-gnu/bin/aarch64-linux-gnu-) BL31=\$(readlink -f out/target/product/rpi4/obj/ATF_OBJ/rpi4/debug/bl31.bin) make -j1 -C external/u-boot O=\$(readlink -f out/target/product/rpi4/obj/UBOOT_OBJ) olddefconfig ) && (PATH=/usr/bin:/bin:\$PATH ARCH=arm64 CROSS_COMPILE=\$(readlink -f prebuilts/gcc/linux-x86/aarch64/gcc-linaro-aarch64-linux-gnu/bin/aarch64-linux-gnu-) BL31=\$(readlink -f out/target/product/rpi4/obj/ATF_OBJ/rpi4/debug/bl31.bin) make -j1 -C external/u-boot O=\$(readlink -f out/target/product/rpi4/obj/UBOOT_OBJ) KCFLAGS=\"-fgnu89-inline \" ) && (cp out/target/product/rpi4/obj/UBOOT_OBJ/u-boot.bin out/target/product/rpi4/obj/UBOOT_OBJ/u-boot.bin.sd )"
TARGET_PRODUCT =  rpi4:
make: Entering directory `/home/m/GloDroid/external/u-boot'
Makefile:37: *** missing separator.  Stop.
make: Leaving directory `/home/m/GloDroid/external/u-boot'
ninja: build stopped: subcommand failed.
10:49:24 ninja failed with: exit status 1

[0;31m#### failed to build some targets (20 seconds) ####[00m

rsglobal, 2020-07-30:
Please share output of the `mount` command also.

myne123, 2020-07-30:
/dev/sda2 on / type ext4 (rw,errors=remount-ro)
proc on /proc type proc (rw,noexec,nosuid,nodev)
sysfs on /sys type sysfs (rw,noexec,nosuid,nodev)
none on /sys/fs/cgroup type tmpfs (rw)
none on /sys/fs/fuse/connections type fusectl (rw)
none on /sys/kernel/debug type debugfs (rw)
none on /sys/kernel/security type securityfs (rw)
none on /sys/firmware/efi/efivars type efivarfs (rw)
udev on /dev type devtmpfs (rw,mode=0755)
devpts on /dev/pts type devpts (rw,noexec,nosuid,gid=5,mode=0620)
tmpfs on /run type tmpfs (rw,noexec,nosuid,size=10%,mode=0755)
none on /run/lock type tmpfs (rw,noexec,nosuid,nodev,size=5242880)
none on /run/shm type tmpfs (rw,nosuid,nodev)
none on /run/user type tmpfs (rw,noexec,nosuid,nodev,size=104857600,mode=0755)
none on /sys/fs/pstore type pstore (rw)
/dev/sda1 on /boot/efi type vfat (rw)
systemd on /sys/fs/cgroup/systemd type cgroup (rw,noexec,nosuid,nodev,none,name=systemd)
gvfsd-fuse on /run/user/1000/gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,user=m)

rsglobal, 2020-08-01:
Thank you for sharing this.
Unfortunately looking into it I have no idea why this happens for you.
Currently @MaksymPrymierov is checking our manual on clean Ubuntu 20.04 in virtual environment, I hope he will also face your issue and can help t resolve it.

myne123, 2020-08-02:
Thanks.
I figured it might be related to the dependencies - so if you can figure out that, then this might go away. 
Which would be nice. 
I'd love to give this a crack on the pi. 

I don't suppose you have a prebuilt image you could post on a torrent or something?

rsglobal, 2020-08-02:
There is a prebuilt images on the github:
https://github.com/GloDroid/glodroid_manifest/releases/tag/v0.3.0

Also you might want to test https://konstakang.com/devices/rpi4/LineageOS17.1/ which is at this moment should be more usable than GloDroid. KonstaKang mixed our experience with UI with his experience with high-level components.
Hope we can enable most of the high level features next release.

myne123, 2020-08-02:
Ah didn't see those. Thanks.
Tried his, it was... ok I guess.

marynasergiyenko, 2020-08-27:
The issue is resolved. Feel free to re-open it if needed.

closed 2020-08-27 by marynasergiyenko