GloDroid/glodroid_manifest #55: RPi4 have no access to sdcard

matviizorin, 2020-07-31:
After changing the directory to `/sdcard` other `adb shell` command we can see the log below:
<details>
  <summary>sdcard</summary>

```
rpi4:/ $ cd sdcard
/system/bin/sh: cd: /sdcard: No such file or directory
```
</details>

sdcard refers to `/mnt/user/0` path (`/sdcard -> /storage/self/primary -> /mnt/user/0/primary`) but containing folder is empty:
<details>
   <summary>folder content</summary>

```
rpi4:/ # ls -la /mnt/user/0
total 0
drwxr-x--x 2 root root 40 1970-01-01 00:00 .
drwxr-xr-x 3 root root 60 1970-01-01 00:00 ..
```
</details>

Also, creating a file on the target device fails with the folowing list of errors in `adb logcat`:

<details>
  <summary>logcat</summary>

```
07-08 09:03:17.299  2160  2179 E DatabaseUtils: Writing exception to parcel
07-08 09:03:17.299  2160  2179 E DatabaseUtils: java.lang.IllegalArgumentException: Parent document isn't a directory
07-08 09:03:17.299  2160  2179 E DatabaseUtils: 	at com.android.internal.content.FileSystemProvider.createDocument(FileSystemProvider.java:239)
07-08 09:03:17.299  2160  2179 E DatabaseUtils: 	at com.android.providers.downloads.DownloadStorageProvider.createDocument(DownloadStorageProvider.java:205)
07-08 09:03:17.299  2160  2179 E DatabaseUtils: 	at android.provider.DocumentsProvider.callUnchecked(DocumentsProvider.java:1121)
07-08 09:03:17.299  2160  2179 E DatabaseUtils: 	at android.provider.DocumentsProvider.call(DocumentsProvider.java:1067)
07-08 09:03:17.299  2160  2179 E DatabaseUtils: 	at android.content.ContentProvider.call(ContentProvider.java:2152)
07-08 09:03:17.299  2160  2179 E DatabaseUtils: 	at android.content.ContentProvider$Transport.call(ContentProvider.java:477)
07-08 09:03:17.299  2160  2179 E DatabaseUtils: 	at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:277)
07-08 09:03:17.299  2160  2179 E DatabaseUtils: 	at android.os.Binder.execTransactInternal(Binder.java:1021)
07-08 09:03:17.299  2160  2179 E DatabaseUtils: 	at android.os.Binder.execTransact(Binder.java:994)
07-08 09:03:17.301  3840  3895 W DocumentsContract: Failed to create document
07-08 09:03:17.301  3840  3895 W DocumentsContract: java.lang.IllegalArgumentException: Parent document isn't a directory
07-08 09:03:17.301  3840  3895 W DocumentsContract: 	at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:170)
07-08 09:03:17.301  3840  3895 W DocumentsContract: 	at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:140)
07-08 09:03:17.301  3840  3895 W DocumentsContract: 	at android.content.ContentProviderProxy.call(ContentProviderNative.java:658)
07-08 09:03:17.301  3840  3895 W DocumentsContract: 	at android.content.ContentProviderClient.call(ContentProviderClient.java:558)
07-08 09:03:17.301  3840  3895 W DocumentsContract: 	at android.content.ContentResolver.call(ContentResolver.java:2032)
07-08 09:03:17.301  3840  3895 W DocumentsContract: 	at android.provider.DocumentsContract.createDocument(DocumentsContract.java:1327)
07-08 09:03:17.301  3840  3895 W DocumentsContract: 	at com.android.documentsui.CreateDirectoryFragment$CreateDirectoryTask.doInBackground(CreateDirectoryFragment.java:151)
07-08 09:03:17.301  3840  3895 W DocumentsContract: 	at com.android.documentsui.CreateDirectoryFragment$CreateDirectoryTask.doInBackground(CreateDirectoryFragment.java:127)
07-08 09:03:17.301  3840  3895 W DocumentsContract: 	at android.os.AsyncTask$3.call(AsyncTask.java:378)
07-08 09:03:17.301  3840  3895 W DocumentsContract: 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
07-08 09:03:17.301  3840  3895 W DocumentsContract: 	at com.android.documentsui.ProviderExecutor.run(ProviderExecutor.java:104)
07-08 09:03:17.303  3840  3895 W Documents: Failed to create directory
07-08 09:03:17.303  3840  3895 W Documents: java.lang.IllegalArgumentException: Parent document isn't a directory
07-08 09:03:17.303  3840  3895 W Documents: 	at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:170)
07-08 09:03:17.303  3840  3895 W Documents: 	at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:140)
07-08 09:03:17.303  3840  3895 W Documents: 	at android.content.ContentProviderProxy.call(ContentProviderNative.java:658)
07-08 09:03:17.303  3840  3895 W Documents: 	at android.content.ContentProviderClient.call(ContentProviderClient.java:558)
07-08 09:03:17.303  3840  3895 W Documents: 	at android.content.ContentResolver.call(ContentResolver.java:2032)
07-08 09:03:17.303  3840  3895 W Documents: 	at android.provider.DocumentsContract.createDocument(DocumentsContract.java:1327)
07-08 09:03:17.303  3840  3895 W Documents: 	at com.android.documentsui.CreateDirectoryFragment$CreateDirectoryTask.doInBackground(CreateDirectoryFragment.java:151)
07-08 09:03:17.303  3840  3895 W Documents: 	at com.android.documentsui.CreateDirectoryFragment$CreateDirectoryTask.doInBackground(CreateDirectoryFragment.java:127)
07-08 09:03:17.303  3840  3895 W Documents: 	at android.os.AsyncTask$3.call(AsyncTask.java:378)
07-08 09:03:17.303  3840  3895 W Documents: 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
07-08 09:03:17.303  3840  3895 W Documents: 	at com.android.documentsui.ProviderExecutor.run(ProviderExecutor.java:104)
```
</details>

rsglobal, 2020-07-31:
v0.3.0 uses kernel v5.7 for broadcom platform.
Starting from kernel v5.6 sdcardfs is not provided by Google, but required for Android up to v10

Best solution for now is to move to v5.4 stable.
Both Google [1] and Broadcom [2] supports it.

[1] - https://android.googlesource.com/kernel/common/+/refs/heads/android-5.4-stable
[2] - https://github.com/raspberrypi/linux/tree/rpi-5.4.y

matviizorin, 2020-07-31:
Thanks, I'm going to change the kernel version in the nearest time

rsglobal, 2020-07-31:
https://github.com/lineage-rpi/android_kernel_brcm_rpi/tree/lineage-17.1

rsglobal, 2020-08-06:
This commits has to be reverted for enabling sdcardfs starting from v5.7. I've successfully done it for PP on kernel v5.8 
```
584c54745dab Revert "ANDROID: vfs: Add setattr2 for filesystems with per mount permissions"
c3271fe2811c Revert "ANDROID: vfs: Add permission2 for filesystems with per mount permissions"
972cde43ff77 Revert "ANDROID: vfs: add d_canonical_path for stacked filesystem support"
e9aaf35681c2 Revert "ANDROID: fs: Restore vfs_path_lookup() export"
a96c98dd7a30 ANDROID: sdcardfs: remove sdcardfs from system
6883a1e06ba8 Revert "ANDROID: fscrypt: add key removal notifier chain"
```

matviizorin, 2020-08-08:
I've got an issue with composition committing after migrating to [v5.8](https://github.com/raspberrypi/linux/tree/rpi-5.8.y) with [android-mainline-tracking](https://android.googlesource.com/kernel/common/+/refs/heads/android-mainline-tracking) patches.

<details>
<summary>logcat</summary>

```
08-07 07:24:25.591   197   197 E hwc-drm-display-compositor: Commit test failed for display 0, FIXME
08-07 07:24:25.591   197   197 E hwc-drm-two: Failed to apply the frame composition ret=-13
08-07 07:24:25.591   205   205 E HWComposer: presentAndGetReleaseFences: present failed for display 0: BadParameter (4)
```
</details>

Trying to investigate and fix this issue.

matviizorin, 2020-08-09:
I've cherry-picked the commit for DRM driver to support Android (https://github.com/GloDroid/glodroid_forks/commit/2fe6160b72530189a0f965eee5435adf86cf350a) and as for now, the composition creation works properly. WIP branch is pushed to the fork as a result ([kernel-broadcom-next](https://github.com/matviizorin/glodroid_forks/commits/broadcom-next)).

rsglobal, 2020-08-12:
Pushed to https://github.com/GloDroid/glodroid_forks/tree/kernel-sunxi-next
Thank you for you efforts.

closed 2020-08-12 by rsglobal