GloDroid/glodroid_manifest #58: Integration builds

rsglobal, 2020-08-07:


rsglobal, 2020-08-07:
Pinephone v1.1 integration build on 7 Aug 2020

New features:
- Kernel v5.8 (With @megous patches)
- GPU offloading in some limited cases.
- Basic bluetooth functionality
- Blinking backlight fixed

Known issues:
- Youtube video playing freezes
- Audio doesn't work

Download: https://drive.google.com/file/d/1BMTsxBWROCh_1OyxssXb3IgQ54hTzjpL/view?usp=sharing

nas00, 2020-08-07:
Pulsing display issue gone
Wifi stability appears improved
Bluetooth can see devices but not pairing so far
Youtube videos only work reliably at 240p and below
No audio at all now

dewitpj, 2020-08-08:
Sorry - no luck with the encrypted password - it's almost as if it forgets the password. The network goes into "Saved" but it keep prompting for the password and never connects

rsglobal, 2020-09-03:
PinePhone integration build updated.

@nas00 ,
Could you test proper selection of PinePhone 1.2 dts, in case you have v1.2?
It can be done by executing `adb shell cat /proc/device-tree/compatible` command.

nas00, 2020-09-03:
I have a braveheart device v1.1 - if you're after ubports edition devices, I can post a call for testers on the pine64 forum?

rsglobal, 2020-09-03:
I will appreciate if you do this. I also own Braveheart v1.1. So it is only feature I can't test.

nas00, 2020-09-04:
@rsglobal is the build compatible with 3gb devices? a user on the forum is having a problem with the flashing script.

rsglobal, 2020-09-05:
@nas00 , It should be compatible. According to logs from forum, there is some problem with script. Or ./fastboot file is corrupted for some reason

Count042, 2020-09-05:
@rsglobal This fails.  This is a separate issue from the one that you addressed earlier with regards to the usb tethering thing.

When I flash the generated image to an sdcard, the pinephone 3gb LED will light up red.  Nothing else happens.  This is the same experience with if I flash either of the provided deploy images.  This was done with the latest integration build.  This also fails when I flash the generated sdcard image that you also suggest trying.  Same result.

If you look here, at the very bottom of the page: https://wiki.pine64.org/index.php?title=PinePhone_Common_Issues

There is a small change required to address the different memory structure of the 3gb Pinephone. 

I will link to the specific commit that enabled this support in uboot when I find it (I'm not sure why they didn't include the link in the pinephone wiki)

heneknenipes, 2020-09-20:
Hi, I have pinephone postmarket edition (I have 3GB and 2GB RAM version). When I install Glodroid v. 0.4.0 - when I press powerbutton the screen is on all the time. Power button doesnt work as it should. How I can fix this? Thanks.

rsglobal, 2020-09-20:
@heneknenipes ,  
Sleep mode is currently broken. You can enable it with reverting commit https://github.com/GloDroid/glodroid_device/commit/90994b291ca55a4d747ecf21ab1b8bf2a5fc3136

Shatur95, 2021-02-21:
@rsglobal, I tested latest `master` (not `next`) with this commit reverted and phone always wakes from sleep.

rsglobal, 2021-03-06:
Latest validation builds uploaded: https://github.com/rsglobal/glodroid_manifest/releases/tag/v0.6.0-validation

squidink7, 2021-03-06:
Just tested that build on my pinephone, and the flash-sd.sh script seems broken. I could only get it to finish by removing the line where it rebooted into fastboot before flashing the a partition, and then it still failed when it came to formatting userdata and metadata to ext4, which I did in gparted manually, now it boots up but is stuck at the android splash screen for about 20 minutes before it turns off.

Quang-vip, 2021-03-08:
@rsglobal I just tested the latest validation build v0.6.0 on the Orange Pi Prime and surprisingly it successfully flashed the system onto the sd card. But then it hung at the android boot screen without any signs of going any further.

Here's what happened in the UART console: https://pastebin.com/AfawK4yH
And on the screen (weird artifacts..):

[image]

Nice work btw! :D

dilinger, 2021-03-10:
> Just tested that build on my pinephone, and the flash-sd.sh script seems broken. I could only get it to finish by removing the line where it rebooted into fastboot before flashing the a partition, [...]

I had the same problem with the 0.6.0 validation build. When the flash-sd.sh script runs './fastboot reboot-fastboot', it reboots into the attached screen. The script hangs forever. I tried the various options, but they don't allow the script to continue.

[image]
[image]

rsglobal, 2021-05-07:
Fixed. Images updated.
https://github.com/rsglobal/glodroid_manifest/releases/tag/v0.6.1-validation

squidink7, 2021-05-07:
I tried flashing it to my pinephone, but the deply-sd.img didn't allow my phone to boot into fastboot. I tried using the deploy-sd.img from the 0.5.1 release which worked but the flashing script got to about halfway through then stopped after 'fastboot oem format' and stayed on 'waiting for any device'

rsglobal, 2021-05-07:
Thank you for reporting this. I do not have such issues on PP1.2. I'll try on v1.1 soon.

squidink7, 2021-05-07:
I'm using 1.2 Manjaro CE, are you trying to use it on your sd card or your emmc?

rsglobal, 2021-05-07:
I was able to reproduce this on my device. Will fix soon. I'm using emmc.

squidink7, 2021-05-08:
wow! that works awesome!! I had to change a line in flash-sd.sh to stop it from trying to reboot directly into fastboot because that caused issues, instead I rebooted manually and went into fastboot when it was done. Other than that amazing release! y'all are brilliant.

nas00, 2021-05-11:
> Fixed. Images updated: 

Is there a new pinetab image available?

rsglobal, 2021-05-12:
https://github.com/rsglobal/glodroid_manifest/releases/download/v0.6.1-validation/v0.6.1_pinetab-userdebug.tar.gz

tophneal, 2021-05-13:
For v0.6.1 on the PineTab:
I couldn't get flash-sd.sh to write to the SD (sd img didn't seem to boot, adb couldn't find the device.) I used the gensdimg.sh and was able to write the resulting img to SD. It does appear to boot (dock keyboard backlights come on, and respond to quick presses of the power button,) but there is no video (or possibly just no backlight) on the display yet.

dilinger, 2021-06-16:
Finally got around to testing this. It failed again, but I managed to work around it. Note that I was using the system fastboot (debian 11), rather than the fastboot from the package.

`Finished. Total time: 0.060s
+ fastboot flash bootloader bootloader-sd.img
Sending 'bootloader' (1920 KB)                     OKAY [  0.140s]
Writing 'bootloader'                               OKAY [  0.125s]
Finished. Total time: 0.280s
+ fastboot flash uboot-env env.img
Sending 'uboot-env' (256 KB)                       OKAY [  0.023s]
Writing 'uboot-env'                                OKAY [  0.023s]
Finished. Total time: 0.062s
+ fastboot reboot
Rebooting                                          OKAY [  0.000s]
Finished. Total time: 0.100s
+ set +x
+ fastboot oem format
< waiting for any device >
                                                   OKAY [  0.011s]
Finished. Total time: 0.011s
+ fastboot flash bootloader bootloader-sd.img
Sending 'bootloader' (1920 KB)                     OKAY [  0.131s]
Writing 'bootloader'                               OKAY [  0.116s]
Finished. Total time: 0.265s
+ fastboot flash uboot-env env.img
Sending 'uboot-env' (256 KB)                       OKAY [  0.023s]
Writing 'uboot-env'                                OKAY [  0.022s]
Finished. Total time: 0.063s
+ fastboot flash recovery_boot boot.img
Sending 'recovery_boot' (28672 KB)                 OKAY [  1.988s]
Writing 'recovery_boot'                            OKAY [  1.497s]
Finished. Total time: 3.514s
+ fastboot erase misc
Erasing 'misc'                                     OKAY [  0.013s]
Finished. Total time: 0.038s
+ fastboot reboot-fastboot
Rebooting into fastboot                            OKAY [  0.005s]
< waiting for any device >
Finished. Total time: 17.196s
+ fastboot flash boot_a boot.img
Sending 'boot_a' (28672 KB)                        OKAY [  2.228s]
Writing 'boot_a'                                   FAILED (Status read failed (No such device))
fastboot: error: Command failed`

I tried rerunning the script a few times, but it would just hang waiting for a device. My pinephone was at the fastboot screen, so I selected "reboot to recovery", and then from there "reboot to fastboot" (or whatever the selection is called), and then:

`dilinger@e7470:~/03-03-2021-15h16m_release_next_pinephone-userdebug/gd-v0.6.1-pinephone-userdebug/artifacts/images$ sudo fastboot flash boot_a boot.img
couldn't parse max-download-size 'no'
Sending 'boot_a' (28672 KB)                        FAILED (Write to device failed in SendBuffer() (Protocol error))
fastboot: error: Command failed
dilinger@e7470:~/03-03-2021-15h16m_release_next_pinephone-userdebug/gd-v0.6.1-pinephone-userdebug/artifacts/images$ sudo fastboot flash boot_a boot.img
< waiting for any device >
Sending 'boot_a' (28672 KB)                        OKAY [  2.597s]
Writing 'boot_a'                                   OKAY [  2.424s]
Finished. Total time: 5.079s
dilinger@e7470:~/03-03-2021-15h16m_release_next_pinephone-userdebug/gd-v0.6.1-pinephone-userdebug/artifacts/images$ sudo fastboot flash dtbo_a boot_dtbo.img
Sending 'dtbo_a' (103 KB)                          OKAY [  0.012s]
Writing 'dtbo_a'                                   OKAY [  0.067s]
Finished. Total time: 0.115s
dilinger@e7470:~/03-03-2021-15h16m_release_next_pinephone-userdebug/gd-v0.6.1-pinephone-userdebug/artifacts/images$ sudo fastboot flash super  super.img

Invalid sparse file format at header magic
Sending sparse 'super' 1/6 (262124 KB)             

OKAY [ 23.699s]
Writing 'super'                                    OKAY [ 19.598s]
Sending sparse 'super' 2/6 (262128 KB)             OKAY [ 23.058s]
Writing 'super'                                    OKAY [ 14.569s]
Sending sparse 'super' 3/6 (262140 KB)             OKAY [ 19.631s]
Writing 'super'                                    OKAY [ 14.585s]
Sending sparse 'super' 4/6 (230452 KB)             OKAY [ 18.594s]
Writing 'super'                                    OKAY [ 13.154s]
Sending sparse 'super' 5/6 (261112 KB)             OKAY [ 21.559s]
Writing 'super'                                    OKAY [ 14.620s]
Sending sparse 'super' 6/6 (19928 KB)              OKAY [  1.511s]
Writing 'super'                                    OKAY [  7.953s]
Finished. Total time: 195.083s
dilinger@e7470:~/03-03-2021-15h16m_release_next_pinephone-userdebug/gd-v0.6.1-pinephone-userdebug/artifacts/images$ 
dilinger@e7470:~/03-03-2021-15h16m_release_next_pinephone-userdebug/gd-v0.6.1-pinephone-userdebug/artifacts/images$ 
dilinger@e7470:~/03-03-2021-15h16m_release_next_pinephone-userdebug/gd-v0.6.1-pinephone-userdebug/artifacts/images$ 
dilinger@e7470:~/03-03-2021-15h16m_release_next_pinephone-userdebug/gd-v0.6.1-pinephone-userdebug/artifacts/images$ 
dilinger@e7470:~/03-03-2021-15h16m_release_next_pinephone-userdebug/gd-v0.6.1-pinephone-userdebug/artifacts/images$ sudo fastboot --wipe-and-use-fbe
Erasing 'userdata'                                 OKAY [  6.258s]
setting FBE marker on initial userdata...
Erase successful, but not automatically formatting.
File system type raw not supported.
Erasing 'metadata'                                 OKAY [  0.019s]
Erase successful, but not automatically formatting.
File system type raw not supported.
Finished. Total time: 6.342s
dilinger@e7470:~/03-03-2021-15h16m_release_next_pinephone-userdebug/gd-v0.6.1-pinephone-userdebug/artifacts/images$ sudo fastboot reboot
Rebooting                                          OKAY [  0.005s]
Finished. Total time: 0.055s`

It then successfully rebooted into glodroid. Thanks!

closed 2022-04-18 by rsglobal