GloDroid/glodroid_manifest #59: PINEPHONE: Implement camera HAL

rsglobal, 2020-08-07:
Unfortunately there is no ready to use camera HAL for Android Q.
At least I could no find one.
We can start adapting `camera.v4l2` from `hardware/libhardware/modules/camera/3_4`, then extend it with flashlight etc.

rsglobal, 2020-08-07:
https://github.com/CoasiaNexell/android_hal_camera/tree/pie-work
Can be also used as reference

rsglobal, 2020-08-10:
More camera HAL which is working for lineage-rpi.
https://osdn.net/projects/android-x86/scm/git/hardware-libcamera/tree/pie-x86/

rsglobal, 2020-08-17:
This HAL seems to be mainline and supported:
https://libcamera.org/docs.html#android-camera-hal

Unfortunately there is no Android.bp/mk exists. Only meason.

kbingham, 2020-10-07:
Indeed we don't yet have an Android blueprint in libcamera, and Android do not support meson or external builds.

We discussed this at the recent Linux plumbers conference, but solutions for integrating meson are not coming soon.

If you construct your own bp integration for Android, the src/android later in libcamera should be already usable, however you might have to add an entry to the simple pipeline handler to support the pinephone.

rsglobal, 2020-10-27:
@kbingham ,

Hi,
Thank you for your help.
I'm just wondering, how do you testing it with Android? Are you building it using NDK?

rsglobal, 2021-06-08:
Initial version available:
https://github.com/kbingham/libcamera/pull/26

rsglobal, 2021-11-11:
Some progress has been done, more info available at : https://github.com/kbingham/libcamera/issues/28

rsglobal, 2021-11-25:
This is already done in GloDroid, but kernel drivers require a few improvements to make it work. See https://github.com/kbingham/libcamera/issues/28 for more info.

closed 2021-11-25 by rsglobal