GloDroid/glodroid_manifest #61: [Info] Create a project status page

hitech95, 2020-08-11:
I would like to know the project status for each board,
Especially about the RPI HW acceleration.

AFIK android works only without the VPU blob (fullkms) , so HW is disabled due to this.
Any plan on testing (firmware kms) to enable the VPU, the ISP and the HW de/encoder?

Camera could work withthe new libcamera, it should also provide an android hal.

rsglobal, 2020-08-11:
@hitech95,
Thank you for your interest in this project.
We will prepare status table shortly.

I am not expert in rpi HW, but as far as I know it has very powerful KMS composer, and it is enabled in GloDroid for UI layers.
And we have a plans to enable it also for Video layers to off-load GPU during video playback.

Video decoding is done by CPU as for now. We do not have and do not want to use any proprietary blobs.
Google promises to create support for v4l2 request API some time ago, and provided a library for h264 decoding. But it doesn't work out-of-the-box.
I hope sooner or later Google or someone else should fix this issue.

I will try enable native USB camera support in the next release.

hitech95, 2020-08-11:
So the problem is still the HAL for codec2 <-> libva
I have an BPI M2M (R16), I will try to build android for that.

Mesa3d should work with that (its a mali GPU) no idea about the hw composer!

matviizorin, 2020-08-14:
> I will try to enable native USB camera support in the next release.

I've successfully implemented it to our BSP by this PR - https://github.com/GloDroid/glodroid_device/pull/10.

Atemu, 2021-12-14:
There seems to be a status entry of sorts on the PinePhone wiki: https://wiki.pine64.org/index.php?title=PinePhone_Software_Releases#GloDroid

herbsmn, 2022-11-04:
The project now has a support matrix for the Pinephone Pro here: https://github.com/GloDroid/glodroid_manifest/issues/221

rsglobal, 2023-04-02:
PP status page is here: https://github.com/GloDroidCommunity/pine64-pinephone/issues/2
PPP status page is here https://github.com/GloDroidCommunity/pine64-pinephonepro/issues/1
RPI4 status page is here: https://github.com/GloDroidCommunity/raspberry-pi/issues/1

closed 2023-04-02 by rsglobal