GloDroid/glodroid_manifest #69: Error checking out repo during sync

davepo, 2020-08-31:
I keep getting this error during the 'repo sync -cq' process:

Aborting
error: prebuilts/ndk/: platform/prebuilts/ndk checkout 258d9d2fa15758c968856e92a6c18908bd841697 
error: Cannot checkout platform/prebuilts/ndk

I am unsure if this is something relative to me or a greater issue with sync setup and/or actual repo.

davepo, 2020-08-31:
Just realized I only posted half of the error message:

Aborting
error: prebuilts/ndk/: platform/prebuilts/ndk checkout 258d9d2fa15758c968856e92a6c18908bd841697 
error: Cannot checkout platform/prebuilts/ndk
Checking out projects:  99% (758/763) platform/tools/tradefederation/contrib^[[1Checking out projects:  99% (762/763), done.

error: Unable to fully sync the tree.
error: Checking out local projects failed.
Failing repos:
prebuilts/ndk

rsglobal, 2020-08-31:
It looks like ndk repository in .repo folder is corrupted.
Could you try to remove all downloaded data and do repo init && repo sync again?

davepo, 2020-09-01:
That worked! Thanks!

closed 2020-09-01 by davepo