GloDroid/glodroid_manifest #70: SIM card usage

swedneck, 2020-09-01:
Is it possible to use the SIM card for mobile data and calling? If so, what do i need to do to enable it?

rsglobal, 2020-09-01:
Hi,
SIM is not yet supported by GloDroid.
Currently you can play with modem only at low level:
https://xnux.eu/devices/feature/modem-pp.html

Supporting it by OS require quectel RIL library at least, and... some additional tuning.

swedneck, 2020-09-01:
Anything (aside from programming) i could to to help bring SIM support?

rsglobal, 2020-09-01:
First step I believe is to insert sim card and make a call using echo ATxxx > /devttyUSB2
After you can confirm it is working, next step is to setup audio paths (to modify XML file).

Remaining steps depends on how much experience in Android development do you have.
I also hope to avoid any programming.

Anyway, at first you have to sync with the next branch. Or I can provide you latest integration build tomorrow.

swedneck, 2020-09-01:
I have 0 experience with android development, but i can certainly test things and report back with results.
How would i run commands? Should i install termux?

rsglobal, 2020-09-02:
I do not have any experience with this modem. Sorry, but I can't help now.
I will back to enabling modem support when I will have got more free time.
So only thing you can help for now is to "dig": i mean read forums, experiment using Linux distros on PP, etc, and sharing the results.

rsglobal, 2020-09-03:
Mobile data support: https://xnux.eu/devices/feature/modem-pp-data.html

swedneck, 2020-09-03:
how would i run commands in android?

rsglobal, 2020-09-03:
1. Use latest integration build or build the next branch. (It includes modem power manager driver module)
https://github.com/GloDroid/glodroid_manifest/issues/58

Use `adb shell` from your HOST PC to open console
`$ su` to enter superuser mode 
After that you should be able to wakeup the modem.

swedneck, 2020-09-03:
i will test!

rsglobal, 2020-09-03:
Android has microcom app, you can try:
```
su
microcom -s 115200 /dev/ttyUSB2
```

rsglobal, 2020-09-03:
This utility can be used to configure qmi:
https://git.openwrt.org/?p=project/uqmi.git;a=summary

@matviizorin , can you add Android.bp to compile it?
Just like https://github.com/rpetrovski/android-x86-external-uqmi for earlier vesion.

matviizorin, 2020-09-07:
@rsglobal 
Yes, I can. I'm going to try it after completing the other available activities.

rsglobal, 2020-09-18:
@bauner is currently WIP on similar activity:
https://github.com/bauner/glodroid_device/commits/ril-gps

rsglobal, 2020-10-25:
Turn on GPS and listen for data:
```
adb shell
$ su
# echo 1 > /sys/class/modem-power/modem-power/device/powered
# cat /dev/ttyUSB1 &
# echo AT+QGPS=1,30,50,0,2 > /dev/ttyUSB2

You should see GPS data on the screen.

To disable GPS:
# echo AT+QGPSEND > /dev/ttyUSB2
```

user293b, 2021-07-14:
Is it realistic that it might work in a future release?

rsglobal, 2021-08-23:
Initial SIM card support is implemented in next branch. It based on Google cuttlefish (emulator) RIL library, therefore isn't 100% compatible with QC25 modem.

closed 2021-08-23 by rsglobal