GloDroid/glodroid_manifest #71: Adb doesn't detect raspberry pi ttl

TheLazyHatGuy, 2020-09-01:
I have tried swapping the RXd and TXD pins with no luck.

Any help is appreciated

TheLazyHatGuy, 2020-09-01:
Got a log using `screen /dev/ttyUSB0 115200`

```
Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 3200
recover4.elf not found
recovery.elf not found
Read start4.elf bytes  2272832 hnd 0x00001bfc hash a439c7fcf5a16d52
Read fixup4.dat bytes     5405 hnd 0x00002d63 hash 75312421da0e036f
0x00c03111 0x00000000 0x0000001f
MEM GPU: 76 ARM: 948 TOTAL: 1024
FIXUP src: 128 256 dst: 948 1024
Starting start4.elf @ 0xfec00200

NOTICE:  BL31: v2.3(debug):de9d0d7c7
NOTICE:  BL31: Built : 22:11:35, Sep  1 2020
INFO:    Changed device tree to advertise PSCI.
INFO:    ARM GICv2 driver initialized
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a72: CPU workaround for 859971 was applied
INFO:    BL31: cortex_a72: CPU workaround for cve_2017_5715 was applied
INFO:    BL31: cortex_a72: CPU workaround for cve_2018_3639 was applied
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x80000
INFO:    SPSR = 0x3c9

U-Boot 2020.07-rc4-ga9a04043fa49-dirty (Sep 01 2020 - 22:11:50 +0100)

DRAM:  3.9 GiB
RPI 4 Model B (0xc03111)
MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
Loading Environment from FAT... *** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
Hit any key to stop autoboot:  0 
ERROR: reserving fdt memory region failed (addr=0 size=80000)
3391 bytes read in 21 ms (157.2 KiB/s)
## Executing script at 03000000
Could not find "misc" partition
** Bad device specification mmc 0#misc **
Couldn't find partition mmc 0#misc

```

rsglobal, 2020-09-02:
Hi,

ADB needs Android running.
You attached log is normal deploy image log. You should be able to flash Android using fastboot at this point. Please refer to https://github.com/GloDroid/glodroid_manifest/blob/master/README.md#step-3

TheLazyHatGuy, 2020-09-02:
It doesn't work. I just get this 

```bash
~/Documents/GloDroid/out/target/product/rpi4$ ./flash-sd.sh 
+ ./fastboot flash gpt deploy-gpt.img
< waiting for any device >
```

rsglobal, 2020-09-02:
Please share your host dmesg log.
Also try to run ./flash-sd.sh with sudo.

TheLazyHatGuy, 2020-09-02:
Sudo is a no go. ADB and Fastboot don't detect a device

```bash
[64764.343424] usb 3-1: USB disconnect, device number 19
[64764.343555] pl2303 ttyUSB0: error sending break = -19
[64764.343638] pl2303 ttyUSB0: pl2303 converter now disconnected from ttyUSB0
[64764.343654] pl2303 3-1:1.0: device disconnected
[64767.734435] usb 3-1: new full-speed USB device number 24 using xhci_hcd
[64767.893412] usb 3-1: New USB device found, idVendor=067b, idProduct=2303, bcdDevice= 4.00
[64767.893413] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[64767.893414] usb 3-1: Product: USB-Serial Controller
[64767.893415] usb 3-1: Manufacturer: Prolific Technology Inc.
[64767.901448] pl2303 3-1:1.0: pl2303 converter detected
[64767.918469] usb 3-1: pl2303 converter now attached to ttyUSB0
[64775.938327] pl2303 ttyUSB0: pl2303_get_line_request - failed: -32
[64775.945346] pl2303 ttyUSB0: pl2303_get_line_request - failed: -32
[64875.878049] usb 3-1: USB disconnect, device number 24
[64875.878167] pl2303 ttyUSB0: error sending break = -19
[64875.878243] pl2303 ttyUSB0: pl2303 converter now disconnected from ttyUSB0
[64875.878251] pl2303 3-1:1.0: device disconnected
[64882.738454] usb 3-1: new full-speed USB device number 25 using xhci_hcd
[64882.897054] usb 3-1: New USB device found, idVendor=067b, idProduct=2303, bcdDevice= 4.00
[64882.897056] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[64882.897056] usb 3-1: Product: USB-Serial Controller
[64882.897057] usb 3-1: Manufacturer: Prolific Technology Inc.
[64882.904085] pl2303 3-1:1.0: pl2303 converter detected
[64882.921119] usb 3-1: pl2303 converter now attached to ttyUSB0
[64896.895897] pl2303 ttyUSB0: pl2303_get_line_request - failed: -32
```

rsglobal, 2020-09-02:
pl2303 seems to be ttl UART converter.
Do you see any dmesg logs related to RPI OTG after boot?
Also ensure your typec designed to transfer data. (I had some problems with OTG that was capable to transfer power only)

rsglobal, 2020-09-02:
Can you also confirm behavior is the same on v0.3.0 release:  
https://github.com/GloDroid/glodroid_manifest/releases/download/v0.3.0/gd-rpi4-v0.3.0.tar.gz

TheLazyHatGuy, 2020-09-02:
```bash
!@pop-os:/$ dmesg | grep otg
!@pop-os:/$ dmesg | grep rpi
!@pop-os:/$ 
```
What do you mean by `ensure your typec designed to transfer data.` I assume the fact I can see the logs proves that data transfer is happening

rsglobal, 2020-09-02:
Hm... You are seeing the logs over ttl-uart converter.
You need to connect RPI typec USB cable, which power-up the board to your host machine.

TheLazyHatGuy, 2020-09-02:
OMG. I just had to use the type-c port on the Pi.
Thanks for your help @rsglobal I feel like an idiot now :D :D

closed 2020-09-02 by TheLazyHatGuy

rsglobal, 2020-09-02:
I glad it worked for you.