GloDroid/glodroid_manifest #74: Try to use Lineage OS manifest and Graphene OS manifest in the same way we are using AOSP now.

rsglobal, 2020-09-03:
If we're lucky, we'll get both OSes with little or no effort.

Lineage manifest: https://github.com/LineageOS/android
Graphene Manifest: https://github.com/GrapheneOS/platform_manifest

kev-dev1, 2020-09-23:
How to build this with Glodroid?

rsglobal, 2020-09-23:
We haven't managed to build it yet.

kev-dev1, 2020-09-23:
Ah okey. Can i try this alone too?

herbsmn, 2020-11-19:
Manifests for the development versions of Replicant 9, 10, and 11 can be found here: https://git.replicant.us/replicant-next/manifest/

Obikawa, 2022-03-15:
Greetings.

How is this integration coming along? My wish is to use Replicant on my Pinephone, which depends on LineageOS.
Is it a matter of  weeks, months, years?

Thank you for your hard work.

Atemu, 2022-03-16:
Android on the Pinephone isn't a thing yet AFAIK. You'll have to wait for the Glodroid efforts; maybe help them financially.

Obikawa, 2022-03-16:
> Android on the Pinephone isn't a thing yet AFAIK. You'll have to wait for the Glodroid efforts; maybe help them financially.

Pardon me, but this is not a question I asked.

Also, while I am open to the idea of patronage or sponsorship, I need to see good communication first.

Obikawa, 2022-03-16:
> Glodroid is the effort to port Android to the Pinephone (and other open platforms). LOS/Replicant support on the Pinephone implies Glodroid.

I know all of this.

I don't know what a "manifest" is, or how to contribute code to Android, etc. The only information I am asking for is "how far away is the current state from being picked up by LineageOS?".

I apologize for being annoying, but there is a language barrier which I want to overcome.

Obikawa, 2022-03-16:
Thank you.

Could I ask one more question? If I wanted to encourage LineageOS to support Pinephone, would this "manifest integration" on GloDroid's side help the case?

While I can see and approve of the progress made by the Linux community on Pine devices, it is still far from being ready for use at work. I could buy an old Replicant-supported phone, but I would rather donate that money towards enablement of the Pinephone.

rsglobal, 2022-03-16:
> Could I ask one more question? If I wanted to encourage LineageOS to support Pinephone, would this "manifest integration" on GloDroid's side help the case?

I have no idea, sorry.

Atemu, 2022-03-16:
> If I wanted to encourage LineageOS to support Pinephone

That's not how it works. LOS is a collective of volunteer hackers who port an Android fork to some of the devices they own.  
Please don't bother any one of them about supporting any new device ever. If they could support it, they would already be doing it; they don't need your encouragement for that.

***

In order to be able to actually port LOS to a device, you need to be intimately familiar with the device.

The only group of people who have the skills and knowledge for the Pinephone I know of are the very people working on Glodroid already. Once AOSP-based Glodroid works, getting also-AOSP-based LOS working could be close to trivial (relatively speaking). In theory that is which is what this issue is about.

If you want LOS on the Pinephone, you want Glodroid on the Pinephone and should support their cause as best as possible. LOS and Replicant support will likely sort themselves out very quickly once Glodroid fully supports it.

Mr-P-Butthole, 2022-12-09:
Seeing as the Pinephone was created as a hobbiest device, its not a huge surprise that its hobbiest supported. I for one applaud your valiant efforts and wish you luck on your mission.

rsglobal, 2023-03-20:
LineageOS preview images available: https://github.com/GloDroidCommunity/pine64-pinephone/pull/10
It has a much better look and feel with the lineage.

LineageOS support for other devices will be added in the same manner.

closed 2023-03-20 by rsglobal