GloDroid/glodroid_manifest #75: Doesn't boot on PinePhone CE (3GB)

colemickens, 2020-09-04:
When I download the images tarball for the pinephone and burn `deploy-sd.img` to an sd card, the phone does not boot or show any signs of life. Removing the SD Card leads to postmarketOS booting from eMMC.

EnderNightLord-ChromeBook, 2020-09-05:
A lot of OS's aren't supported for the 3GB version, you pretty much gotta wait for the patch to come through all the other operating systems before you can do much.

colemickens, 2020-09-05:
My understanding is that the Glodroid SD Card image for the pinephone would contain the uboot. I'm not sure I understand how other distros updating will help with the glodroid_manifests?

Or does glodroid pull from a manjaro uboot fork, or is my understanding totally off?

Thanks for your time.

rsglobal, 2020-09-05:
@colemickens ,

deploy-sd.img is not an Android image, it only enables fastboot USB gadget. Please complete step 3 of readme:
https://github.com/GloDroid/glodroid_manifest#step-3

In case you are using v0.3.0, consider using latest integration build:
https://github.com/GloDroid/glodroid_manifest/issues/58#issuecomment-670354906

Count042, 2020-09-05:
@rsglobal This fails. This is a separate issue from the one that you addressed earlier with regards to the usb tethering thing.

When I flash the generated image to an sdcard, the pinephone 3gb LED will light up red. Nothing else happens. This is the same experience with if I flash either of the provided deploy images. This was done with the latest integration build. This also fails when I flash the generated sdcard image that you also suggest trying. Same result.

If you look here, at the very bottom of the page: https://wiki.pine64.org/index.php?title=PinePhone_Common_Issues

There is a small change required to address the different memory structure of the 3gb Pinephone.

I will link to the specific commit that enabled this support in uboot when I find it (I'm not sure why they didn't include the link in the pinephone wiki)

Count042, 2020-09-05:
ah, finding the commit was as easy as searching for the string specified in the wiki on Google rather then DuckDuckGo:
http://u-boot.10912.n7.nabble.com/RFC-PATCH-sunxi-support-asymmetric-dual-rank-DRAM-on-A64-R40-td416975.html

rsglobal, 2020-09-06:
Thank you for pointing to this commit.
I've added it to the build. Can you check if it works for you?
https://drive.google.com/file/d/1xfAF_zbvTr8uK8a03h4ej-gM8tt_xByD/view?usp=sharing

Count042, 2020-09-06:
Thank you for adding that so quickly!   That seems to have done the trick.  I'm booting into Android for the first time now of off the eMMC!

jpmeijers, 2020-09-06:
@rsglobal it works for me, thanks.

rsglobal, 2020-09-06:
@Count042 ,

Thank you for testing.
Can you execute `adb shell cat /proc/device-tree/compatible` at host and share result?

Count042, 2020-09-06:
@rsglobal pine64,pinephone-1.2allwinner,sun50i-a64⏎

jpmeijers, 2020-09-06:
Also `pine64,pinephone-1.2allwinner,sun50i-a64` here.

rsglobal, 2020-09-06:
Thanks for testing.
Fix is included into the next branch and will be available in the next release.

closed 2020-09-06 by rsglobal

ropij, 2020-09-07:
I'm having trouble getting either of the deploy images to boot on my 3GB CE PinePhone, even with the build posted here. I'm unsure if it's related or not to this but I've got this UART log:
```
U-Boot SPL 2020.07-rc4-00051-g565d50f55c0c-dirty (Sep 06 2020 - 09:46:12 +0300)
DRAM: 3072 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.3(debug):de9d0d7c7
NOTICE:  BL31: Built : 11:11:42, Aug 18 2020
NOTICE:  BL31: Detected Allwinner A64/H64/R18 SoC (1689)
NOTICE:  BL31: Found U-Boot DTB at 0x40990d0, model: Pine64 PinePhone (1.2)
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP803 on RSB
INFO:    PMIC: dcdc1 voltage: 3.300V
INFO:    PMIC: dcdc5 voltage: 1.200V
INFO:    PMIC: dcdc6 voltage: 1.100V
INFO:    PMIC: dldo1 voltage: 3.300V
INFO:    PMIC: dldo2 voltage: 1.800V
INFO:    PMIC: dldo3 voltage: 2.800V
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 843419 was applied
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
NOTICE:  PSCI: System suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9

U-Boot 2020.07-rc4-00051-g565d50f55c0c-dirty (Sep 06 2020 - 09:46:12 +0300) Allwinner Technology

CPU:   Allwinner A64 (SUN50I)
Model: Pine64 PinePhone (1.2)
DRAM:  3 GiB
MMC:   Device 'mmc@1c11000': seq 1 is in use by 'mmc@1c10000'
mmc@1c0f000: 0, mmc@1c10000: 2, mmc@1c11000: 1
Loading Environment from FAT... Unable to use mmc 1:1... In:    serial@1c28000
Out:   serial@1c28000
Err:   serial@1c28000
Allwinner Sun4I USB PHY driver loaded
sun4i_usb_phy_xlate: phy_id = 0
Allwinner mUSB OTG (Peripheral)
Net:   
Warning: usb_ether MAC addresses don't match:
Address in ROM is		de:ad:be:ef:00:01
Address in environment is	02:ba:e6:57:52:36
eth0: usb_ether
Hit any key to stop autoboot:  2  1  0 
3952 bytes read in 3 ms (1.3 MiB/s)
## Executing script at 53000000
Could not find "misc" partition
** Bad device specification mmc 0#misc **
Couldn't find partition mmc 0#misc
starting USB...
Bus usb@1c1a000: USB EHCI 1.00
Bus usb@1c1a400: USB OHCI 1.0
Bus usb@1c1b000: sun4i_usb_phy_xlate: phy_id = 1
USB EHCI 1.00
Bus usb@1c1b400: sun4i_usb_phy_xlate: phy_id = 1
USB OHCI 1.0
scanning bus usb@1c1a000 for devices... 1 USB Device(s) found
scanning bus usb@1c1a400 for devices... 1 USB Device(s) found
scanning bus usb@1c1b000 for devices... 1 USB Device(s) found
scanning bus usb@1c1b400 for devices... 1 USB Device(s) found
sun4i_usb_phy_xlate: phy_id = 0
sun4i_usb_phy_vbus_detect: id_det = -22
```
This is the dmesg output:
```
[10587.352539] usb 1-1: new high-speed USB device number 11 using xhci_hcd
[10587.501077] usb 1-1: New USB device found, idVendor=18d1, idProduct=4ee0, bcdDevice= 2.15
[10587.501079] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[10587.501081] usb 1-1: Product: USB download gadget
[10587.501083] usb 1-1: Manufacturer: Allwinner Technology
[10587.501084] usb 1-1: SerialNumber: [redacted]
```

The eMMC is flashed with Mobian nightly if that's important.

rsglobal, 2020-09-07:
@ropij ,

According to your logs, deploy image started successfully. Have you tried to proceed installation using ./flash-sd.sh ?

ropij, 2020-09-07:
I'm dense, I never tried running ./flash-sd.sh with elevated privileges. It seems to be working and I'll edit this post with the results.

rsglobal, 2020-09-07:
`sudo` is not required if `adb` package is installed.