GloDroid/glodroid_manifest #76: Default "web browser" can't download files

jpmeijers, 2020-09-06:
Using the Pinephone CE.

The default web browser is just a webview with a URL bar. It does not seem like it can download files. Also USB file sharing doesn't work. It makes it therefore very difficult to get an APK onto the phone.

How about adding Firefox or Chromium or another browser by default?

rsglobal, 2020-09-06:
We will add one soon
Currently you can use `adb push` to upload the files.

rsglobal, 2020-09-06:
Thank you for your request.
We are working on this.

jholtiv, 2020-09-17:
@jpmeijers Try using the duckduckgo browser from F-Droid until it's working

osakanataro, 2020-09-27:
@jpmeijers other way, you can use  Bromite(based on chromium browser) via F-Droid.
But Bromite not include default F-Droid repository. 
Add bromite repository https://fdroid.bromite.org/fdroid/repo on F-Droid settings 
details https://www.bromite.org/fdroid

rsglobal, 2020-10-02:
Bromite looks like a good choice. Thank you

rsglobal, 2021-05-06:
Mozilla Firefox (fenix) installed.