GloDroid/glodroid_manifest #77: Install gapps

jpmeijers, 2020-09-06:
Can one install OpenGapps onto glodroid? What would the process be?

https://opengapps.org/

kev-dev1, 2020-09-15:
I tried to flash the OpenGapps TV-Stock but it's failed. How can i flash the OpenGapps

g-lebars, 2020-11-10:
I was able to modify the build process for the opi3 by following the  tutorial at: https://github.com/opengapps/aosp_build
There is however some changes that need to be done:
1. the `repo forall -c git lfs pull` fails due to opencl.
You can however call 
`repo forall vendor/opengapps/sources/* -c git lfs pull`

2. You need to set:
`BOARD_SUPER_PARTITION_SIZE := 2147483648 
BOARD_GLODROID_DYNAMIC_PARTITION_SIZE := 2143289344`
in `device/glodroid/common/boardconfig-common.mk` to accomodate for the added apps.

However I still struggle to push the super partition to the board.

When I call flash-sd.sh the script fails at 
`./fastboot flash super super.img` 
with 
`Writing 'super'                                    FAILED (remote: 'Operation not permitted')`

rsglobal, 2020-11-10:
Hi,

You also have to update partition layout:
https://github.com/GloDroid/glodroid_device/blob/1fe99ca9dd51119b07ce1e162b040b673f7c22fe/platform/uboot/bootscript.cpp#L36

g-lebars, 2020-11-11:
Perfect thank you.
And thanks to the developer who did a wonderful job

kev-dev1, 2020-11-14:
Work´s Andorid with GApps

ekpyron, 2020-11-14:
@kev-dev1 Yes, I at least have a working custom build of Glodroid v0.5.1 with working GApps on a Pinephone following the process described by @g-lebars in https://github.com/GloDroid/glodroid_manifest/issues/77#issuecomment-724945350 together with changing https://github.com/GloDroid/glodroid_device/blob/1fe99ca9dd51119b07ce1e162b040b673f7c22fe/platform/uboot/bootscript.cpp#L36 to ``EXTENV(partitions, ";name=super,size=2048M,uuid=\${uuid_gpt_super}")`` as suggested by @rsglobal in https://github.com/GloDroid/glodroid_manifest/issues/77#issuecomment-724986371.
(While on the other hand trying to flash prebuilt opengapps by adb sideloading into the prebuilt Glodroid v0.5.1 release fails some signature verification for me with the few Android 11 builds of opengapps I found and doesn't seem to be expected to work without modifications in general IIUC.)

kev-dev1, 2020-11-14:
Ok. Thanks for the Information!

g-lebars, 2020-11-15:
I was able to build and flash the rom with aosp 11 and opengapps to an opi3.
However there are still problems. because the board was not validated on google side.
On first boot you will have graphical errors during the initialization if you dont disable wifi and bluetooth. 
You can then go through the device settings to set the wifi.
when the board is set google protect will complain about system verification. 
To mitigate this problem you can use the steps described here: https://www.xda-developers.com/google-blocks-gapps-uncertified-devices-custom-rom-whitelist/
But even with this your access to google play will be limited. Many apps will not be available due to the weak play protect validation.
I'm looking into magisk to help with that but for now I was unable to flash it on the opi 3

ekpyron, 2020-11-15:
@g-lebars Yes, I had similar issues, although for me hitting skip fast enough during wiki setup in device initialization was enough.
I also had to whitelist the device like https://www.xda-developers.com/google-blocks-gapps-uncertified-devices-custom-rom-whitelist/, but I thought that was to be expected.
Regarding the remaining issues of play protect validation: I'm also not able to e.g. flash magisk and it doesn't look like they make it easy to include magisk in the aosp build - ideally I'd like to generate custom signing keys to the glodroid build and use them to sign things like magisk images, but I have yet to find out where exactly to put such keys, s.t. they will end up in glodroid recovery...

rsglobal, 2023-05-02:
Tesla Android project uses GloDroid with Gapps on top. Please find a reference at https://github.com/tesla-android/android-raspberry-pi

closed 2023-05-02 by rsglobal