GloDroid/glodroid_manifest #78: KMS-based video playback offload playing very smooth, but has some visible shade issue

rsglobal, 2020-09-07:
Hope to fix this issue by setting correct color_encoding and color_range .
See more at:
https://gitlab.freedesktop.org/drm-hwcomposer/drm-hwcomposer/-/issues/38

matviizorin, 2020-09-07:
I've started to investigate it. Many thanks for the effort

rsglobal, 2020-09-08:
I've checked dataspace value in both videos that has visible shades mismatches.
It was 0x10C10000, which corresponds to:
https://cs.android.com/android/platform/superproject/+/master:system/core/libsystem/include/system/graphics-base-v1.0.h;l=93;drc=7d940ae4afa450696afa25e07982f3a95e17e9b2

rsglobal, 2020-09-08:
After additional investigation, I want to make some adjustments to this issue:

1. Issue is not visible to the naked eye. Color shades difference between standards are not so big.
2. Issue that was observed while playing video in Gallery app was due to some on-hover event highlight, and only visible when mouse cursor are used.
3. Supporting for different color standards in HWC still relevant.

matviizorin, 2020-09-08:
I've just created the draft MR ([!115](https://gitlab.freedesktop.org/drm-hwcomposer/drm-hwcomposer/-/merge_requests/115)).

rsglobal, 2020-09-08:
@matviizorin ,

Thank you for your effort. Let's continue discussion in the MR.

closed 2020-09-08 by rsglobal