GloDroid/glodroid_manifest #79: question: can you build on a arm64 device?

purringChaos, 2020-09-09:
The only devices I have right now are pine arm64 devices and I wanted to build the very latest version for pinetab which isn't on the releases tab.
If anyone can quickly send me a build of pinetab glodroid that'd be great.

rsglobal, 2020-09-09:
Hi,

Try this one: https://drive.google.com/file/d/1xip8gvOiGRR5-42n4z7OUySE9bSvqNI2/view?usp=sharing

tophneal, 2020-09-12:
> Hi,
> 
> Try this one: https://drive.google.com/file/d/1xip8gvOiGRR5-42n4z7OUySE9bSvqNI2/view?usp=sharing

Thanks for providing that. However, I just used it to create a bootable SD for my PineTab, and it seems to suffer from the same issue as some other pre release images for PineTab: the OS boots, but it does not display video.

rsglobal, 2020-10-02:
Issue is about PineTab images. They are available under assets tab in the Releases page:
https://github.com/GloDroid/glodroid_manifest/releases
Closing.

PS: We will fix pinetab build when get real device.

closed 2020-10-02 by rsglobal