GloDroid/glodroid_manifest #80: Pinephone no GPS, no network location

jpmeijers, 2020-09-11:
I'm testing an app on GloDroid on the Pinephone.

The following code checks if GPS and network locations providers are available before enabling them.

```
        // Register the listener with the Location Manager to receive location updates
        if (locationManager!!.isProviderEnabled(LocationManager.GPS_PROVIDER) ) {
            locationManager!!.requestLocationUpdates(LocationManager.GPS_PROVIDER, 0L, 0f, locationListener)
        }
        if (locationManager!!.isProviderEnabled(LocationManager.NETWORK_PROVIDER)) {
            locationManager!!.requestLocationUpdates(LocationManager.NETWORK_PROVIDER, 0L, 0f, locationListener)
        }
```

Both GPS_PROVIDER and NETWORK_PROVIDER are false. This means the app can not obtain the phone's location.

rsglobal, 2020-09-19:
@jpmeijers ,

You can try @bauner 's work adding GSM&GSP:
https://forum.pine64.org/showthread.php?tid=10613&pid=78278#pid78278

rsglobal, 2020-10-25:
GPS RE:
https://gist.github.com/alastair-dm/263209b54d01209be28828e555fa6628

rsglobal, 2023-01-21:
This issue is tracked now by https://github.com/GloDroid/glodroid_manifest/issues/104#issuecomment-1399277765.

closed 2023-01-21 by rsglobal