GloDroid/glodroid_manifest #82: [Question] How to update without data lost.

kev-dev1, 2020-09-15:
How to update or upgrade the newest build without to lost the Data.

rsglobal, 2020-09-17:
Hi @kev-dev1 ,

It depends on many factors.
The main is partition layout.
In case new release has the same partition layout as previous, you can just put a comment on `./fastboot format userdata & metadata` lines in `./flash.sh` script and flash.

In theory this should work for v0.3.0 to v0.4.0 transition, but nobody from GD team tested it. So there is a risk for you to lose your data.

To eliminate any risks mobile brands have test department, which tests how will OTA influence user data and system stability  before publishing release.

kev-dev1, 2020-09-17:
Ok. I will test it in the next Days and see if it is possible to update without data loss

kev-dev1, 2020-09-20:
The Upgrade from version 0.3.0 to 0.4.0 works.

closed 2020-09-20 by kev-dev1