GloDroid/glodroid_manifest #84: rpi4: "sgdisk" is not allowed to be used

dmitvitalii, 2020-09-18:
### Environment
```
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=10
TARGET_PRODUCT=rpi4
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a53
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=cortex-a15
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-5.4.0-26-generic-x86_64-Ubuntu-18.04.5-LTS
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=QQ3A.200605.002.A1
OUT_DIR=out
```

### Steps

- `. build/envsetup.sh`
- `lunch rpi4-userdebug`
- `make images`

### Result
_"sgdisk" is not allowed to be used. See https://android.googlesource.com/platform/build/+/master/Changes.md#PATH_Tools for more information._

### Workaround
`make TEMPORARY_DISABLE_PATH_RESTRICTIONS=true  images`

### Possible Solution
Either use another tools, listed in `build/soong/ui/build/paths/config.go`, or add `"sgdisk": Allowed` to `Configurations` map in that file.

rsglobal, 2020-09-18:
Hi @dmitvitalii ,

Thank you for your report. Can you execute `$ which sgdisk` from the terminal and share result?

dmitvitalii, 2020-09-18:
**Before sourcing && lunch**
`/sbin/sgdisk`
**After sourcing && lunch**
`/aosp/out/soong/host/linux-x86/bin/sgdisk`

**OS**
Ubuntu-18.04.5-LTS using Docker

rsglobal, 2020-09-18:
To overcome such AOSP restriction we are modifying PATH variable:
https://github.com/GloDroid/glodroid_device/blob/6ad27a5047990119f297fdaef2a10bcfba4b43f9/platform/tools/tools.mk#L3
It should also work in your case. We never used docker to build, so it might be a reason.

dmitvitalii, 2020-09-18:
Thank you for pointing it up, I will try to find the reason. After the first successful build, this issue no longer appears.

rsglobal, 2020-10-02:
You're welcome. In case issue is still relevant please reopen it.

closed 2020-10-02 by rsglobal