GloDroid/glodroid_manifest #85: gbm_gralloc: Implement validateBufferSize() method

rsglobal, 2020-09-19:
To fix VtsHalGraphicsMapperV2_1TargetTest test:
https://cs.android.com/android/platform/superproject/+/master:hardware/interfaces/graphics/mapper/2.1/vts/functional/VtsHalGraphicsMapperV2_1TargetTest.cpp;l=117;drc=49f262471d8456ee3e421ea112c9c34dda58774f

Method description is here:
https://cs.android.com/android/platform/superproject/+/master:hardware/libhardware/include/hardware/gralloc.h;l=333;drc=5543b1eb045a4946449982de99474a441fdc1e9d

matviizorin, 2020-11-10:
PR has been created (https://github.com/GloDroid/glodroid_forks/pull/8). The changes should be upstreamed after our project acceptance.

matviizorin, 2020-11-25:
I've just created a PR to the upstream repository (https://github.com/robherring/gbm_gralloc/pull/19).

rsglobal, 2020-12-08:
Thanks. Closing as done.

closed 2020-12-08 by rsglobal