GloDroid/glodroid_manifest #86: PineTab Blank Screen

squierts, 2020-09-19:
I'm able to flash the sd image to the PineTab, but the screen remains blank even after the reboot following a successful flash. I've tried an external monitor with no luck. The logs seemed to indicate it did fully boot.

rsglobal, 2020-09-25:
@Icenowy ,  

Hi, 

Do you have a chance to check it and share the logs?

rsglobal, 2020-09-28:
Hi @squierts ,

Thank you for opening this issue.

Please run the following commands as soon as PineTab boots and attach the results to this issue:

```
adb root
adb logcat | tee logcat.txt
adb shell dmesg | tee dmesg.txt 
adb shell lsmod | tee lsmod.txt
adb shell lshal | tee lshal.txt
adb shell lsof | tee lsof.txt
```

tophneal, 2020-09-29:
Attached are the requested logs, using the latest version, v0.5.1
[dmesg.txt](https://github.com/GloDroid/glodroid_manifest/files/5300206/dmesg.txt)
[logcat.txt](https://github.com/GloDroid/glodroid_manifest/files/5300207/logcat.txt)
[lshal.txt](https://github.com/GloDroid/glodroid_manifest/files/5300208/lshal.txt)
[lsmod.txt](https://github.com/GloDroid/glodroid_manifest/files/5300209/lsmod.txt)
[lsof.txt](https://github.com/GloDroid/glodroid_manifest/files/5300210/lsof.txt)

rsglobal, 2020-09-29:
Thank you,

It looks like back-light driver isn't enabled.

LouisDK1, 2020-11-07:
A similar issues appeared with Postmaket OS.

Maybe this is relevant: https://gitlab.com/postmarketOS/pmaports/-/merge_requests/1627

rakor, 2020-11-28:
The display of the PineTab changed short before it was shipped. That was the reason why at first the official PineTab images did not work. The backgroud light seems to switch on but the screen stays black.

nicodt, 2020-12-10:
Hi I will also test I have the same issue where the display is just white and sleeps after a few seconds and goes blank. I am connected via adb shell though.

rsglobal, 2021-05-25:
My pinetab is working with v0.6.1 . Discord users were reporting pinetab was working as well . There are still a lot of issues with it, we are waiting for fixes from pine64 community.

closed 2021-05-25 by rsglobal

rakor, 2021-05-28:
tried 6.1 on sd with my pinetab. Screen stays black here.

jkoan, 2021-07-02:
> tried 6.1 on sd with my pinetab. Screen stays black here.

I just install GloDroid on PineTab as well, and can confirm that `deploy-sd.img` the screen stays black, but when looking at dmesg (on Host) you will see somthint like this:
```
[122104.161085] usb 1-1.2: New USB device found, idVendor=18d1, idProduct=4ee0, bcdDevice= 2.15
[122104.161092] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[122104.161095] usb 1-1.2: Product: USB download gadget
[122104.161098] usb 1-1.2: Manufacturer: Allwinner Technology
[122104.161100] usb 1-1.2: SerialNumber: 92c0eababc8699d8
```

at this moment you can start the `flash-sd.sh` script.
Of course sd can be substituted with emmc