GloDroid/glodroid_manifest #87: [Raspberry Pi4] Bluetooth not enable/not working

kev-dev1, 2020-09-22:
Bluetooth on Raspberry Pi4 not working to enable. Can we add Bluetooth for Raspberry Pi4.

danialbehzadi, 2021-06-07:
Repeat this one. This should not be so hard, since [konstakang](https://github.com/lineage-rpi/android_kernel_brcm_rpi) and [arpi](https://github.com/android-rpi/kernel_arpi) kernels support Bluetooth well.

danialbehzadi, 2021-06-08:
Adding `BT_BCM` to kernel config seems to be the Bluetooth magic: https://github.com/lineage-rpi/android_kernel_brcm_rpi/commit/4a2059b200425c15e3a590d874cd447d1bf16054

SimoneFrau, 2021-11-15:
Hi,
I've tried to enable bluetooth on the [next](https://github.com/GloDroid/glodroid_manifest/tree/next) branch but without success. These are the steps I've performed:

- kernel/glodroid-broadcom/arch/arm64/configs/bcm2711_config updated with changes described for lineageos_rpi4_defconfig adding various `CONFIG_BT`...
- device/glodroid/rpi4/device.mk updated calling `bluetooth.mk` instead of `no-bluetooth.mk`
- common/bluetooth/boardconfig.mk updated adding first only `BOARD_HAVE_BLUETOOTH_BCM`

but when I try to enable bluetooth, I get several fails and moreover using `adb shell dumpsys bluetooth_manager` name and address are always null. I have then tried other changes:

- device/glodroid/common/bluetooth/boardconfig.mk updated adding also `BOARD_BLUETOOTH_BDROID_BUILDCFG_INCLUDE_DIR` and `BOARD_CUSTOM_BT_CONFIG` but this means adding the folder with files as for android-rpi with `bdroid_buildcfg.h`, `bt_vendor.conf` and `vnd_rpi4.txt`too
- device/glodroid/common/bluetooth/bluetooth.mk updated with bt_vendor.conf:system/etc/bluetooth/bt_vendor.conf and also android.hardware.bluetooth@1.0-impl and android.hardware.bluetooth@1.1-service.btlinux or alternatively using 1.0-service and 1.0-impl but with the same result

I ultimately changed also device/glodroid/common/init.common.rc moving and changing the rfkill management as for android-rpi but without success.

My apologies for the request, I know you said it should not be so hard, probably I am just missing something more trivial which is not present in the GloDroid tree. I've not changed anything more than what I have just described here. Thank you for your support!

rsglobal, 2021-11-19:
I've submitted a minor fix that should help debugging it further:
https://android-review.googlesource.com/c/platform/system/bt/+/1896708

rsglobal, 2021-11-19:
Ok, here is my current progress on bluetooth. All necessary kernel modules is up, but still I can't see no devices to pair.

```diff
Author: Roman Stratiienko <r.stratiienko@gmail.com>
Date:   Fri Nov 19 16:50:39 2021 +0200

    WIP: rpi4: Enable bluetooth
    
    1. Switch serial port to UART1 (PL011)
    2. Enable in-kernel bluetooth on UART0.
    
    Signed-off-by: Roman Stratiienko <r.stratiienko@gmail.com>

diff --git a/rpi4/BoardConfig.mk b/rpi4/BoardConfig.mk
index 275509e..d7b519d 100644
--- a/rpi4/BoardConfig.mk
+++ b/rpi4/BoardConfig.mk
@@ -24,4 +24,4 @@ BOARD_MESA3D_GALLIUM_DRIVERS := vc4 v3d
 BOARD_MESA3D_VULKAN_DRIVERS := broadcom
 
 BOARD_KERNEL_CMDLINE += coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 \
-                        snd_bcm2835.enable_headphones=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000 console=ttyAMA0,115200
+                        vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000 console=ttyS0,115200
diff --git a/rpi4/config.txt b/rpi4/config.txt
index 464d7c2..b222466 100644
--- a/rpi4/config.txt
+++ b/rpi4/config.txt
@@ -1,4 +1,6 @@
 dtparam=audio=on
+dtparam=krnbt=on
+dtparam=uart1=on
 
 [all]
 start_x=1
@@ -11,7 +13,6 @@ armstub=bl31.bin
 camera_auto_detect=1
 
 dtoverlay=dwc2
-dtoverlay=uart0
 dtoverlay=vc4-kms-v3d
 
 max_framebuffers=2
diff --git a/rpi4/device.mk b/rpi4/device.mk
index 9bab512..ffb791e 100644
--- a/rpi4/device.mk
+++ b/rpi4/device.mk
@@ -3,7 +3,7 @@
 # Copyright (C) 2020 Roman Stratiienko (r.stratiienko@gmail.com)
 
 $(call inherit-product, device/glodroid/common/device-common.mk)
-$(call inherit-product, device/glodroid/common/bluetooth/no-bluetooth.mk)
+$(call inherit-product, device/glodroid/common/bluetooth/bluetooth.mk)
 
 # Firmware
```

SimoneFrau, 2021-11-19:
Interesting, did you not add BCM4345C0.hcd?
I just followed all the steps in android-rpi:

https://github.com/android-rpi/device_arpi_rpi4/commit/dde16ec3f1b4f60ecb5c4c40de4f68c75e9ed23b

When you do adb shell dumpsys bluetooth_manager
can you see the name and the address?

SimoneFrau, 2021-11-19:
I see only the name, but the address is always null. These are interesting things I can see from my log:

...
11-19 15:36:40.238  2620  2646 I bt_btif_core: btif_init_bluetooth entered
11-19 15:36:40.239  2620  2646 I bt_stack_config: init attempt to load stack conf from /etc/bluetooth/bt_stack.conf
11-19 15:36:40.240  2620  2652 I droid.bluetoot: [1119/153640.240118:INFO:message_loop_thread.cc(175)] Run: message loop starting for thread bt_jni_thread
11-19 15:36:40.240  2620  2646 I bt_btif_core: btif_init_bluetooth finished
11-19 15:36:40.240  2620  2646 I bt_stack_manager: event_init_stack finished
11-19 15:36:40.241  2620  2620 I bt_osi_wakelock: wakelock_set_os_callouts set to non-native
11-19 15:36:40.241  2620  2620 I bt_btif : get_profile_interface: id = socket
11-19 15:36:40.242  2620  2652 E bt_btif_storage: btif_storage_get_adapter_property: Controller not ready! Unable to return Bluetooth Address
11-19 15:36:40.242  2620  2652 E BluetoothServiceJni: adapter_properties_callback: Status 1 is incorrect
...
...
hwservicemanager: Since android.hardware.bluetooth@1.1::IBluetoothHci/default is not registered, trying to start it as a lazy HAL.
11-19 15:36:44.631  2620  2664 I HidlServiceManagement: getService: Trying again for android.hardware.bluetooth@1.1::IBluetoothHci/default...
11-19 15:36:44.837   187  2671 W libc    : Unable to set property "ctl.interface_start" to "android.hardware.bluetooth@1.1::IBluetoothHci/default": error code: 0x20
11-19 15:36:45.311  2620  2650 F libc    : Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 2650 (alarm_default_c), pid 2620 (droid.bluetooth)
...

rsglobal, 2021-11-19:
I've got some initial initialization but it crashes soon with:
```
10-07 14:48:42.845  2422  2466 F bt_stack: [FATAL:btm_ble_gap.cc(541)] Check failed: p_vcs_cplt_params->param_len >= BTM_VSC_CHIP_CAPABILITY_RSP_LEN. 
```

rsglobal, 2021-11-19:
> Interesting, did you not add BCM4345C0.hcd?

No, thanks. Let me try.

rsglobal, 2021-11-19:
It seems to work:

```
Bluetooth Status
  enabled: true
  state: ON
  address: DC:A6:32:0F:C3:90
  name: rpi4
  time since enabled: 00:00:10.673
```

rsglobal, 2021-11-19:
It still crashes during pairing, but at least there is a progress.

SimoneFrau, 2021-11-19:
Nice! But then or I'm missing some kernel modules or I'm doing something wrong somewhere else because I still do not see the address. Can you tell me which CONFIG_BT... you put on bcm2711_config and defconfig?

rsglobal, 2021-11-19:
I haven't enabled any extra configs.
Just that patch (next branch was updated).
Ensure you've executing `m images`, not just `m` before flashing.

rsglobal, 2021-11-19:
Here is my lsmod

```
Module                  Size  Used by
snd_soc_hdmi_codec     20480  2 
ov5647                 20480  2 
v3d                    86016  13 
gpu_sched              45056  1 v3d
hci_uart               49152  0 
bcm2835_v4l2           45056  0 
btbcm                  20480  1 hci_uart
videobuf2_vmalloc      20480  1 bcm2835_v4l2
bluetooth             479232  3 hci_uart,btbcm
ecdh_generic           16384  1 bluetooth
ecc                    45056  1 ecdh_generic
joydev                 32768  0 
snd_bcm2835            28672  0 
bcm2835_codec          49152  0 
bcm2835_isp            32768  5 
bcm2835_mmal_vchiq     36864  3 bcm2835_v4l2,bcm2835_codec,bcm2835_isp
bcm2835_unicam         49152  3 
vc_sm_cma              36864  2 bcm2835_isp,bcm2835_mmal_vchiq
v4l2_dv_timings        36864  1 bcm2835_unicam
vc4                   286720  51 
v4l2_fwnode            24576  2 ov5647,bcm2835_unicam
v4l2_mem2mem           40960  1 bcm2835_codec
cec                    57344  1 vc4
raspberrypi_hwmon      16384  0 
videobuf2_dma_contig    24576  3 bcm2835_codec,bcm2835_isp,bcm2835_unicam
drm_kms_helper        245760  2 vc4
i2c_mux_pinctrl        16384  0 
videobuf2_v4l2         32768  5 bcm2835_v4l2,bcm2835_codec,bcm2835_isp,bcm2835_unicam,v4l2_mem2mem
videobuf2_common       61440  6 bcm2835_v4l2,bcm2835_codec,bcm2835_isp,bcm2835_unicam,v4l2_mem2mem,videobuf2_v4l2
drm                   573440  20 v3d,gpu_sched,vc4,drm_kms_helper
i2c_mux                16384  1 i2c_mux_pinctrl
i2c_brcmstb            20480  0 
videodev              458752  15 ov5647,bcm2835_v4l2,bcm2835_codec,bcm2835_isp,bcm2835_unicam,v4l2_fwnode,v4l2_mem2mem,videobuf2_v4l2,videobuf2_common
drm_panel_orientation_quirks    20480  1 drm
brcmfmac              294912  0 
snd_soc_core          233472  2 snd_soc_hdmi_codec,vc4
brcmutil               20480  1 brcmfmac
videobuf2_memops       20480  2 videobuf2_vmalloc,videobuf2_dma_contig
i2c_bcm2835            16384  0 
snd_compress           24576  1 snd_soc_core
cfg80211              847872  1 brcmfmac
snd_pcm_dmaengine      20480  1 snd_soc_core
sysimgblt              16384  1 drm_kms_helper
syscopyarea            16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
fb_sys_fops            20480  1 drm_kms_helper
backlight              20480  1 drm
rpivid_mem             16384  0 
nvmem_rmem             16384  0 
uio_pdrv_genirq        16384  0 
uio                    24576  1 uio_pdrv_genirq
```

SimoneFrau, 2021-11-19:
I got it! And I'm pairing without crashes! We have Bluetooth up & running here!

Now I do not have time to put all the changes, I have to clean a bit because I was trying a lot of things but on Monday I will post everything.

Thank you very much for the help and enjoy your weekend!

rsglobal, 2021-11-19:
I suspect I used wrong firmware.
Now I am trying official: https://github.com/RPi-Distro/bluez-firmware/blob/e7fd166981ab4bb9a36c2d1500205a078a35714d/broadcom/BCM4345C0.hcd

rsglobal, 2021-11-19:
Much stable now. I was able to pair my phone. Next branch updated.

closed 2021-11-19 by rsglobal